Livox-SDK/livox_ros_driver #1: Error:   Received an invalid TCPROS header.  Each line must have an equals sign.

crisscrosskao, 2019-09-16:
I am running the `roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list="0TFDFCE00505101"` from my Nvidia Xavier on ROS Melodic and Ubuntu 18.04, on an Arm64 processor.

I connected the Livox by Ethernet to my Xavier computer and confirmed that both my computer and Livox Mid-40 have static IP addresses. I also confirmed previously that I was able to see the Livox lidar when I used it separately on my Ubuntu 16.04 setup on my virtual machine on my Mac via VMWare.

Here, I saw Rviz open up but with no data points. I also couldn't test the Livox Viewer since I cannot open it, probably because I am on a computer with Arm64 processor. The error output from the Livox viewer was the following:
```
christopherkao@christopherkao-xavier:~/Downloads/Livox Viewer For Linux Ubuntu16.04_x64 0.5.0$ livox_viewer.sh: 11: livox_viewer.sh: ./so/livox_viewer: Exec format error
^C
```

Here is the output when I tried to run the `roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list="0TFDFCE00505101"`:
```
christopherkao@christopherkao-xavier:~/livox_ws/src/ws_livox$ roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list:="0TFDFCE00505101"
... logging to /home/christopherkao/.ros/log/acd7bbd6-d848-11e9-a972-00044bcba063/roslaunch-christopherkao-xavier-7378.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://christopherkao-xavier:43629/

SUMMARY
========

PARAMETERS
 * /livox_msg_type: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    livox_lidar_publisher (livox_ros_driver/livox_lidar_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [7391]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to acd7bbd6-d848-11e9-a972-00044bcba063
process[rosout-1]: started with pid [7405]
started core service [/rosout]
process[livox_lidar_publisher-2]: started with pid [7412]
[ INFO] [1568614230.064185685]: Livox-SDK ros demo
[ INFO] [1568614230.064931554]: broadcast code list :000000000000001
[ INFO] [1568614230.065042537]: Invalid bd:000000000000001
[ INFO] [1568614230.065137455]: Commandline input 0TFDFCE00505101
[ INFO] [1568614230.065205939]: commandline input bd:0TFDFCE00505101
[ INFO] [1568614230.065286680]: list all valid bd:
[ INFO] [1568614230.065401183]: 0TFDFCE00505101
[ INFO] [1568614230.089117477]: Publish PointCloud2
process[rviz-3]: started with pid [7415]
[2019-09-15 23:10:31.071] [console] [info]  Boradcast broadcast code: 0TFDFCE00505101  [device_discovery.cpp] [OnBroadcast] [161]
[ INFO] [1568614231.071927816]: Receive Broadcast Code 0TFDFCE00505101, please add it to broacast_code_list if want to connect!

[2019-09-15 23:10:31.072] [console] [info]  LocalIP: 192.168.1.20  [device_discovery.cpp] [OnBroadcast] [213]
[2019-09-15 23:10:31.072] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [214]
[2019-09-15 23:10:31.073] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [215]
[2019-09-15 23:10:31.095] [console] [info] New Device  [device_discovery.cpp] [OnData] [110]
[2019-09-15 23:10:31.095] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [111]
[2019-09-15 23:10:31.095] [console] [info] Broadcast Code: 0TFDFCE00505101  [device_discovery.cpp] [OnData] [112]
[2019-09-15 23:10:31.095] [console] [info] Type: 1  [device_discovery.cpp] [OnData] [113]
[2019-09-15 23:10:31.095] [console] [info] IP: 192.168.1.80  [device_discovery.cpp] [OnData] [114]
[2019-09-15 23:10:31.095] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [115]
[2019-09-15 23:10:31.095] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [116]
[ INFO] [1568614231.096706414]: OnDeviceChange broadcast code 0TFDFCE00505101 update type 0
[ INFO] [1568614231.096946877]: Device State status_code 0
[ INFO] [1568614231.097053284]: Device State working state 5
[ INFO] [1568614231.097144873]: Device feature 0
[2019-09-15 23:10:31.097] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [233]
[2019-09-15 23:10:31.119] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [104]
[ INFO] [1568614231.119764076]: firm ver: 3.2.0.0
[2019-09-15 23:10:31.143] [console] [info]  Update State to 1, device connect true  [device_manager.cpp] [UpdateDeviceState] [224]
[ INFO] [1568614231.143560343]: OnDeviceChange broadcast code 0TFDFCE00505101 update type 2
[ INFO] [1568614231.143746595]: Device State status_code 0
[ INFO] [1568614231.145343748]: Device State working state 1
[ INFO] [1568614231.146114547]: Device feature 0
[2019-09-15 23:10:31.146] [console] [info]  Send Command: Set 0 Id 4 Seq 5  [command_channel.cpp] [Send] [233]
[2019-09-15 23:10:31.167] [console] [info]  Recieve Ack: Set 0 Id 4 Seq 5  [command_handler.cpp] [OnCommand] [104]
[ INFO] [1568614231.167695031]: OnSampleCallback statue 0 handle 0 response 0
[ INFO] [1568614231.193830961]: packet loss : 1689189507410
Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp
```

crisscrosskao, 2019-09-16:
Based on my reading on TCPROS (http://wiki.ros.org/ROS/TCPROS), it appears that ROS uses TCPROS for "ROS Messages and Services." So most likely it may be possible that one of the messages being sent by the Livox SDK Driver might be having a problem with the header, where "each line must have an equals sign."

crisscrosskao, 2019-09-16:
Update: I had both Livox-SDK-ROS and livox_ros_driver in my /src folder in my `livox_ws`. I had built them together before. 

I removed Livox-SDK-ROS, since it seems that that one will be deprecated in favor of livox_ros_driver. Then I ran `catkin_make` again from the `livox_ws` directory. Now I only see the error once, and instead I see about once a second INFO statements of the form

```
[INFO] ... In automatic connection mode, will connect 0TF...
Boradcast (btw that's misspelled in the code) broadcast code...
Receive broadcast code. Please add it to broacast (that's also misspelled) if you want to connect!
```

And it just keeps looping that. I truncated some of the text above. But in Rviz I still don't see anything and a `rostopic echo /livox/lidar` also doesn't show any messages coming out.

crisscrosskao, 2019-09-16:
So I ended up getting the Livox-SDK-ROS to work. I also get the same ERROR message there but in that case it shows messages in Rviz.

Also I do see rod topics coming in when I use Livox-SDK-ROS for the /livox/lidar. I noticed that I don't see any point clouds in Rviz until I uncheckmark "Unreliable." I feel like by default that should be unchecked.

crisscrosskao, 2019-09-16:
Also just mentioning a few tips for others who may get stuck:
1. The Livox-SDK-ROS package is already organized like a workspace. When you run catkin_make it puts /devel and /build folders in that package. I had made a workspace called livox_ws and then placed Livox-SDK-ROS inside /livox_ws/src/. But oddly that package then put /livox_ws/src/Livox-SDK-ROS/build and /devel, etc. So make sure you source devel/setup.bash from inside Livox-SDK-ROS.
2. You need to look inside the main.cpp file and include your Mid-40's serial number inside the broadcast_code_list array. This is not specified in the README.

Livox-SDK, 2019-09-16:
hi,crisscrosskao
   It‘s recommended to use livox_ros_driver package, and please refer to the readme file in livox_ros_driver;
(1)
install Livox-SDK firstly，please
refer to：
https://github.com/Livox-SDK/Livox-SDK/
(2)   
git clone https://github.com/Livox-SDK/livox_ros_driver.git ws_livox/src
(3)
cd ws_livox
(5)
catkin_make

Livox-SDK, 2019-09-16:
For ubuntu18.04, you shuld uncheck the "Unreliable" in rviz, when the "Received an invalid TCPROS header. " error occured!
Thanks for your advice, crisscrosskao! We'll fix this bug soon!

Livox-SDK, 2019-09-16:
crisscrosskao，
   Our livox_rso_driver can run in TX2 and Xavier， but on ubuntu18.04 in Xavier, you shuld uncheck the "Unreliable" in rviz.
   Thanks again crisscrosskao!

Livox-SDK, 2019-09-16:
Hi,crisscrosskao:
   The rviz config file have already updated,git pull or clone a new copy please!
    Thanks a lot!

closed 2019-10-15 by Livox-SDK