Livox-SDK/livox_ros_driver #5: Installation on Win10

mi4869, 2019-11-28:
Hi,

Is there any installation guidelines for Win10? I encountered a problem when building livox_ros_driver package:

NMAKE : fatal error U1077: “"E:\Visual Studio\VC\Tools\MSVC\14.23.28105\bin\Hostx64\x64\cl.exe"”: Return code“0x2” Stop.
NMAKE : fatal error U1077: “"E:\Visual Studio\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"”: Return code“0x2” Stop.
NMAKE : fatal error U1077: “"E:\Visual Studio\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"”: Return code“0x2” Stop.
Invoking "nmake" failed

mi4869, 2019-11-28:
I think I find a reason. There is no unistd.h file under Windows. I tried some ways to use "include#<io.h> and <process.h>" or "#include <windows.h>", but still fail to invoke nmake. Any solutions to install livox_ros_driver on Win10?

Livox-SDK, 2019-12-16:
Sorry,The livox_ros_driver don't support windows!

closed 2020-05-07 by Livox-SDK

runeg96, 2020-09-19:
I know this Issue was closed a long time ago but I made the livox_ros_driver work on ROS melodic Windows 10. If anybody is interested the livox_ros work on Windows 10 see my [repository](https://github.com/runeg96/livox_ros_driver) 

@mi4869 If you still need it could maybe be of some use?

I also made a small guide on how to install it, which is a bit more complicated :D