Livox-SDK/livox_ros_driver #6: How to get the Lidar caliberation parameters into ROS

nithi89, 2019-12-06:
The original Livox visualization software has a feature to do extrinsic calibration of the lidar, which can be stored to the Lidar itself.

When I try to receive the lidar information in ROS, the calibration doesn't seem to take effect and each time, the orientation of each lidar changes arbitrarily.

How can I get calibrated Lidar data from the Livox via ROS?

Livox-SDK, 2019-12-16:
The new version（v2.0.0） of livox_ros_driver support the extrinsic, it will be released in next two week!

nithi89, 2019-12-17:
Is it possible to get it earlier, we have a deadline to meet in the coming week!

Livox-SDK, 2019-12-23:
hi,nithi89
  The new version of livox_ros_driver is released , you can update it now.

nithi89, 2020-01-06:
Thank you guys!

closed 2020-05-07 by Livox-SDK