Livox-SDK/livox_ros_driver #8: timestamp表示什么时间？

Leon1128, 2019-12-27:
您好！我需要对Livox mid-40与其他传感器(camera or IMU)的数据进行时间同步，其他传感器数据的时间戳一般是unix time（从1970-01-01至今的秒数），但是livox发布的数据的时间戳比较特殊，导致无法进行同步。livox驱动中的时间戳是这样的：
timestamp = GetStoragePacketTimestamp(&storage_packet, data_source);
cloud.header.stamp = ros::Time(timestamp / 1000000000.0);  // to ros time stamp
我不太明白GetStoragePacketTimestamp获取的时间代表什么含义呢？麻烦您解答一下，谢谢！

griz1112, 2020-01-02:
I'm having the same issue. Trying to input pointclouds from the livox into gmapping and hector_slam. ROS complains that the timestamps are too old and discards them.

Livox-SDK, 2020-01-03:
/** Timestamp sync mode define. */
typedef enum {
  kTimestampTypeNoSync = 0, /**< No sync signal mode. */
  kTimestampTypePtp = 1,    /**< 1588v2.0 PTP sync mode. */
  kTimestampTypeRsvd = 2,   /**< Reserved use. */
  kTimestampTypePpsGps = 3, /**< pps+gps sync mode. */
  kTimestampTypePps = 4,    /**< pps only sync mode. */
  kTimestampTypeUnknown = 5 /**< Unknown mode. */
} TimestampType;
你好，livox 设备一共支持5种时间戳格式，默认没有外接同步信号的情况下，输出的时间戳是第一种格式，即从设备上电开始的绝对时间，单位是ns

Livox-SDK, 2020-01-03:
如果需要和别的设备进行同步，需要给mid-40输入pps同步信号

Livox-SDK, 2020-01-03:
GetStoragePacketTimestamp获取的时间代表以太网数据包中第一个点云数据的时间，该时间已经转换为绝对时间，单位是ns

Leon1128, 2020-01-04:
好的，明白啦，谢谢！

closed 2020-01-04 by Leon1128

YangSiri, 2021-04-01:
HI @Livox-SDK , 请问可以直接使用 livox_msg.header.stamp = ros::Time::now();吗？

Vineet-Pandey, 2021-04-12:
Hi @Livox-SDK- HOw do we ensure that the timestamp of the ros message published is the UNix time and not the relative time? TIA,