Livox-SDK/livox_ros_driver #12: Build fail

Karltlbm, 2020-02-18:
Hello,

I am having trouble to build the driver.
Althought I have ros installed on my machine, the library can't be included.
Do you have any idea ?
```
/dev/ws_livox/src/livox_ros_driver/livox_ros_driver/lddc.cpp:39:10: fatal error: pcl_ros/point_cloud.h: No such file or directory
 #include <pcl_ros/point_cloud.h>
```

Livox-SDK, 2020-02-19:
Hi，Karltlbm
It is recommended to install the ros full version！
For example,install ros melodic in ubuntu18.04:
sudo apt install ros-melodic-desktop-full

samsdolphin, 2020-03-14:
> Hi，Karltlbm
> It is recommended to install the ros full version！
> For example,install ros melodic in ubuntu18.04:
> sudo apt install ros-melodic-desktop-full

This works for me!

closed 2020-05-07 by Livox-SDK