Livox-SDK/livox_ros_driver #14: failed to load lvx file and show in rviz

lyh2833495, 2020-02-27:
Hi,

when I run the command "roslaunch livox_ros_driver lvx_to_rosbag_rviz.launch" throw the issue:

...
[INFO] [1582801801.888035630]: livox/imu publish imu data, Publisher QueueSize[4096]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
=======
REQUIRED process [livox_lidar_publisher-1] has died!
process has died [pid 7458, exit code -6, cmd /home/user/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/lyh/.ros/log/a2764ea2-5942-11ea-8291-000c29292ea9/livox_lidar_publisher-1.log].
log file: /home/user/.ros/log/a2764ea2-5942-11ea-8291-000c29292ea9/livox_lidar_publisher-1*.log
Initiating shutdown!
=======
[record-2] killing on exit
[livox_lidar_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Don't know how to fix it 

Thank you

Livox-SDK, 2020-03-02:
What is the firmware version of your horizon? 
You can stop publishing imu packets  for testing.
Set the parameter "imu_rate" to '0', which is in livox_lidar_config.json or livox_hub_config.json file, the livox_ros_driver will read it and tell the horizon to stop publishing imu packets.

lyh2833495, 2020-03-02:
It works!

Thank you!

I just use ros to load the lvx file on official website ,without using real lidar

closed 2020-05-07 by Livox-SDK