Livox-SDK/livox_ros_driver #18: Failed to convert the lvx data file to rosbag file

InitialWang, 2020-03-19:
Hi,
I excute the follow command roslaunch livox_ros_driver lvx_to_rosbag.launch lvx_file_path:="xxxxxx".
Throw the issue:

> Start to read lvx file.
[ INFO] [1584608695.672558908]: Support only one topic.
[ INFO] [1584608695.673374797]: livox/lidar publish use PointCloud2 format, publisher queue size [4096]
Read progress : 1
Read progress : 2
Read progress : 3
Read progress : 4
Read progress : 5
Read progress : 6
Read progress : 7
Read progress : 8
Read progress : 9
Read progress : 10
Read progress : 11
Read progress : 12
Read progress : 13
Read progress : 14
Read progress : 15
Read progress : 16
Read progress : 17
Read progress : 18
Read progress : 19
Read progress : 20
Read progress : 21
Read progress : 22
Read progress : 23
Read progress : 24
Read progress : 25
Read progress : 26
Read progress : 27
Read progress : 28
Read progress : 29
Read progress : 30
Read progress : 31
Read progress : 32
Read progress : 33
Read progress : 34
Read progress : 35
Read progress : 36
Read progress : 37
Read progress : 38
Read progress : 39
Read progress : 40
Read progress : 41
Read progress : 42
Read progress : 43
Read progress : 44
Read progress : 45
Read progress : 46
Read progress : 47
Read progress : 48
Read progress : 49
Read progress : 50
[ INFO] [1584608696.135814489]: Support only one topic.
[ INFO] [1584608696.136708602]: livox/imu publish imu data, Publisher QueueSize[4096]
terminate called after throwing an instance of 'std::runtime_error'
what(): Time is out of dual 32-bit range
================================================================================REQUIRED process [livox_lidar_publisher-2] has died!
process has died [pid 20133, exit code -6, cmd /home/vic/catkin_ws/src/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/vic/.ros/log/b35834fe-69c0-11ea-bbe7-b42e992833a3/livox_lidar_publisher-2.log].
log file: /home/vic/.ros/log/b35834fe-69c0-11ea-bbe7-b42e992833a3/livox_lidar_publisher-2*.log
Initiating shutdown!

[livox_lidar_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

The point_cloud_data.lvx file was downloaded from [https://www.livoxtech.com/downloads].
May I have your help to fixed it.
Thank you!

InitialWang, 2020-03-19:
I reboot my pc, throw a new issue:

> process[livox_lidar_publisher-2]: started with pid [4080]
[ INFO] [1584609969.157452670]: Livox Ros Driver Version: 2.0.0
[ INFO] [1584609969.161318494]: Data Source is lvx file.
[ INFO] [1584609969.161862372]: Create bag file :/home/vic/Downloads/Sample Point Cloud of Horizon/point_cloud_data.bag!
Livox file version[1]
LvxFile[/home/vic/Downloads/Sample Point Cloud of Horizon/point_cloud_data.lvx] have 3 lidars
[ INFO] [1584609969.161929528]: Init lds lvx file success!
Start to read lvx file.
[ INFO] [1584609969.163333399]: Support only one topic.
[ INFO] [1584609969.163811688]: livox/lidar publish use PointCloud2 format, publisher queue size [4096]
Read progress : 1 
[ INFO] [1584609969.172361772]: Support only one topic.
[ INFO] [1584609969.173117769]: livox/imu publish imu data, Publisher QueueSize[4096]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
================================================================================REQUIRED process [livox_lidar_publisher-2] has died!
process has died [pid 4080, exit code -6, cmd /home/vic/catkin_ws/src/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/vic/.ros/log/aa6cb420-69c3-11ea-a75e-b42e992833a3/livox_lidar_publisher-2.log].
log file: /home/vic/.ros/log/aa6cb420-69c3-11ea-a75e-b42e992833a3/livox_lidar_publisher-2*.log
Initiating shutdown!
================================================================================
[livox_lidar_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Livox-SDK, 2020-03-20:
Pull this project and update the code, checkout the branch draft_pr, and try again!

closed 2020-05-07 by Livox-SDK