Livox-SDK/livox_ros_driver #21: build failed in livox_ros_driver/lddc.cpp

cshan, 2020-04-06:
Hi,

I am build livox ros driver on my raspberry pi, it show me these error:

home/pi/ws_livox/src/livox_ros_driver/livox_ros_driver/lddc.cpp:31:10: fatal error: 'pcl_ros/point_cloud.h' file not found
#include <pcl_ros/point_cloud.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
make[2]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:158: livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:469: livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all] Error 2
make: *** [Makefile:158: all] Error 2

I am already installed pcl 1.7.2 and ros kinetic.

Thanks

xmba15, 2020-04-11:
pcl_ros is the interface of pcl on ros that comes from pcl_perception.
https://github.com/ros-perception/perception_pcl
I think you first should check if your system has this library with
```bash
dpkg -l | grep perception_pcl
```
also livox_ros_driver misses to add pcl_ros dependency inside CMakeLists.txt and package.xml that might cause this library to fail to find pcl_ros.
I can make a PR if the library owner is OK with the change.

Livox-SDK, 2020-04-17:
Sure，thanks！

cshan, 2020-04-19:
@xmba15 Thanks, this changes will be helpful.

closed 2020-04-29 by Livox-SDK