Livox-SDK/livox_ros_driver #22: behavior when launching multiple sensors (without hub)

xmba15, 2020-04-06:
Hi, 
I tried to run the two horizon lidars with 
```bash
roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list:="broadcast_lidar1&broadcast_lidar2"
```
I could see only one /livox/lidar pointcloud message? is it the correct behavior?
Should I expect to see two separate point cloud messages?

In the case of only one point cloud message, does that mean the ros package conduct the point cloud fusion internally?

I did not use livox hub for this experiment.
Thanks for the support.

Livox-SDK, 2020-04-08:
（1）replace broadcast_lidar1 or broadcast_lidar1  with your lidar broadcast code（15 char） ，
         for example， your lidar broadcast code is “0TFDG3B006H2Z11“，the command should be：
        roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list:="0TFDG3B006H2Z11”

Livox-SDK, 2020-04-08:
（2）You can use：
 roslaunch livox_ros_driver livox_lidar_rviz.launch
The ros driver will connect LiDAR uint(s) automatically

closed 2020-04-08 by Livox-SDK