Livox-SDK/livox_ros_driver #27: How to create a depth image ?

Tariq-Abuhashim, 2020-05-12:
The majority of object segmentation methods utilise a depth image in their learning and inference.
Is there a way to specify the dimensions of the depth image from livox horizon pointcloud data?
Is there a method to create depth image in the SDK?

Livox-SDK, 2020-05-12:
Sorry, Tariq-Abuhashim , our SDK don't support it

closed 2020-07-31 by Tariq-Abuhashim

xmba15, 2020-08-08:
@Tariq-Abuhashim I actually created a library here that supports spherical projection of livox point cloud data into 2d image here.
https://github.com/xmba15/livox_lidar_data_manipulation

xmba15, 2020-08-08:
[image]
here is the result sample