Livox-SDK/livox_ros_driver #28: 驱动的接口文档说明

xielinjiang, 2020-05-15:
你好：
     请问你们有ros驱动接口的文档说明吗，我指的是驱动里面一些函数的说明文档。
谢谢！盼复。

Livox-SDK, 2020-05-16:
只有Livox-SDK 的接口说明文档；
livox ros dirver 对外publish pointcloud2格式的点云数据，应用端节点只需订阅该topic即可；

xielinjiang, 2020-05-18:
我是想参照 livox_ros_driver 实现一个不基于ros的livox驱动，你们有类似的demo吗？

Livox-SDK, 2020-05-18:
可以参考下Livox-SDK 下sample_cc的demo实现

xielinjiang, 2020-05-18:
```c++
void LdsLidar::OnDeviceBroadcast(const BroadcastDeviceInfo *info) {
  if (info == nullptr) {
    return;
  }

  if (info->dev_type == kDeviceTypeHub) {
    printf("In lidar mode, couldn't connect a hub : %s\n", info->broadcast_code);
    return;
  }

  if (g_lidars->IsAutoConnectMode()) {
    printf("In automatic connection mode, will connect %s\n", info->broadcast_code);
  } else {
    if (!g_lidars->FindInWhitelist(info->broadcast_code)) {
      printf("Not in the whitelist, please add %s to if want to connect!\n",\
             info->broadcast_code);
      return;
    }
  }

  bool result = false;
  uint8_t handle = 0;
  result = AddLidarToConnect(info->broadcast_code, &handle);
  if (result == kStatusSuccess && handle < kMaxLidarCount) {
    SetDataCallback(handle, LdsLidar::GetLidarDataCb, (void *)g_lidars);
    LidarDevice* p_lidar = &(g_lidars->lidars_[handle]);
    p_lidar->handle = handle;
    p_lidar->connect_state = kConnectStateOff;
    p_lidar->config.enable_fan = true;
    p_lidar->config.return_mode = kStrongestReturn;
    p_lidar->config.coordinate = kCoordinateCartesian;
    p_lidar->config.imu_rate = kImuFreq200Hz;
  } else {
    printf("Add lidar to connect is failed : %d %d \n", result, handle);
  }
}
```
你说的是这个回调函数吗？每次回调只返回一个点云吗？每秒钟能扫描出多少个点，这个数量是固定的吗？

Livox-SDK, 2020-05-18:
sample_cc/lidar 中：
void LdsLidar::GetLidarDataCb(uint8_t handle, LivoxEthPacket *data,
                              uint32_t data_num, void *client_data) {
或者ros driver的lds_lidar.cpp 源文件中：
void LdsLidar::OnLidarDataCb(uint8_t handle, LivoxEthPacket *data,
                             uint32_t data_num, void *client_data) {

每次接收一个完整的以太网包，每个以太网包中点云的个数在用户手册中有说明：
mid40/mid100 每个以太网中包含100个点，没秒总点数100K；
Horizon/tele 每个以太网中包含96个点，每秒总点数是240K；

xielinjiang, 2020-05-18:
sample_cc/lidar 中：
```
void LdsLidar::GetLidarDataCb(uint8_t handle, LivoxEthPacket *data,
uint32_t data_num, void *client_data)
```
LivoxEthPacket 就是一个以太网包吗？

```
typedef struct {
  uint8_t version;              /**< Packet protocol version. */
  uint8_t slot;                 /**< Slot number used for connecting LiDAR. */
  uint8_t id;                   /**< LiDAR id. */
  uint8_t rsvd;                 /**< Reserved. */
  uint32_t err_code;      /**< Device error status indicator information. */
  uint8_t timestamp_type;       /**< Timestamp type. */
  /** Point cloud coordinate format, refer to \ref PointDataType . */
  uint8_t data_type;
  uint8_t timestamp[8];         /**< Nanosecond or UTC format timestamp. */
  uint8_t data[1];              /**< Point cloud data. */
} LivoxEthPacket;
```
按照您上面说的，每个以太网包中包含100个点，意思是函数
```
void LdsLidar::GetLidarDataCb(uint8_t handle, LivoxEthPacket *data,
uint32_t data_num, void *client_data)
```
的`LivoxEthPacket *data`里面有100个点？

Livox-SDK, 2020-05-19:
是的

xielinjiang, 2020-05-19:
如上所说：
```c++
void LdsLidar::GetLidarDataCb(uint8_t handle, LivoxEthPacket *data,
                              uint32_t data_num, void *client_data) {
  using namespace std;

  LdsLidar* lidar_this = static_cast<LdsLidar *>(client_data);
  LivoxEthPacket* eth_packet = data;

  if (!data || !data_num || (handle >= kMaxLidarCount)) {
    return;
  }

  if (eth_packet) {
    lidar_this->data_recveive_count_[handle] ++;
    if (lidar_this->data_recveive_count_[handle] % 100 == 0) {
      printf("receive packet count %d %d\n", handle, lidar_this->data_recveive_count_[handle]);

      /** Parsing the timestamp and the point cloud data. */
      uint64_t cur_timestamp = *((uint64_t *)(data->timestamp));
      if(data ->data_type == kCartesian) {
        LivoxRawPoint *p_point_data = (LivoxRawPoint *)data->data;
      }else if ( data ->data_type == kSpherical) {
        LivoxSpherPoint *p_point_data = (LivoxSpherPoint *)data->data;
      }else if ( data ->data_type == kExtendCartesian) {
        LivoxExtendRawPoint *p_point_data = (LivoxExtendRawPoint *)data->data;
      }else if ( data ->data_type == kExtendSpherical) {
        LivoxExtendSpherPoint *p_point_data = (LivoxExtendSpherPoint *)data->data;
      }else if ( data ->data_type == kDualExtendCartesian) {
        LivoxDualExtendRawPoint *p_point_data = (LivoxDualExtendRawPoint *)data->data;
      }else if ( data ->data_type == kDualExtendSpherical) {
        LivoxDualExtendSpherPoint *p_point_data = (LivoxDualExtendSpherPoint *)data->data;
      }else if ( data ->data_type == kImu) {
        LivoxImuPoint *p_point_data = (LivoxImuPoint *)data->data;
      }
    }
  }
}
```

对于函数 `GetLidarDataCb`中，参数`LivoxEthPacket *data`表示一个以太网络包，里面包含100个激光点云数据。那么我该如何获取  `LivoxEthPacket *data`中的100个点数据呢？

Livox-SDK, 2020-05-19:
这只是一个简单的范例，例子中只取了一个点的数据，目的是示范如何从以太网包中取点；
可的数据处理，ros driver先将整个以太网包缓存到环形队列中；然后在另外一个单独的线程中读取并解析数据 以参考下ros driver 中

xielinjiang, 2020-05-19:
```c++
      /** Parsing the timestamp and the point cloud data. */
      uint64_t cur_timestamp = *((uint64_t *)(data->timestamp));
      std::cout << "timestamp:" << cur_timestamp << std::endl;
      if(data ->data_type == kCartesian) { // 笛卡尔坐标点云
      LivoxRawPoint *p_point_data = (LivoxRawPoint *)data->data;
        for (int ii = 0; ii < 100; ++ii) {
          std::cout << "p_point_data: "<< cnt << ", [" << p_point_data->x/1000.0 << "," << p_point_data->y/1000.0 << "," << p_point_data->z/1000.0 << "," << +p_point_data->reflectivity << "]" << std::endl;
          p_point_data++;
        }
      }
```
是不是这样子访问的？

Livox-SDK, 2020-05-19:
是的

xielinjiang, 2020-05-19:
好的，非常感谢

closed 2020-05-19 by xielinjiang