Livox-SDK/livox_ros_driver #29: how to fusion point cloud from hub

lyh2833495, 2020-05-29:
How to fusion point cloud from hub, can't  find "import Extrinsics Tool" and "Start Cali" button in Livox Viewer 0.7.0

Livox-SDK, 2020-06-01:
please reference the Livox Viewer User Manual:
[Livox Viewer User Manual](https://www.livoxtech.com/3296f540ecf5458a8829e01cf429798e/downloads/Livox%20Viewer/Livox%20Viewer%20User%20Manual.pdf)
on page 11

xmba15, 2020-06-08:
@Livox-SDK Please correct me if I am wrong; but I think the way to fuse pointcloud mentioned in Livox Viewer User Manual only applying rotation and translation transformation of the point cloud based on an initial guess. That will introduce error that can integrated over time; hence not suitable way for task that needs precision like SLAM. More consideration after initial guess of transformation should be applying point cloud registration.

Livox-SDK, 2020-06-08:
Hi, xmba15, your suggestion is very useful. In subsequent versions, we may add an automatic calibration function according to different scenarios( with or without overlap).

xmba15, 2020-06-12:
@Livox-SDK thanks, for the reply. Just want to share my experience, but the case of less overlap might be challenging. You might need to implement livox pointcloud specific registration algorithm for the case of less/no overlapping case. My quick test with icp/ndt from PCL did not work so well.

closed 2020-09-07 by Livox-SDK