Livox-SDK/livox_ros_driver #31: 在线切换激光雷达的运行状态

xielinjiang, 2020-06-02:
您好：
    我可以在驱动运行的时候接受指令来动态地切换激光雷达的运行状态吗？
     状态具体指的是：
```
/** Lidar mode. */
typedef enum {
  kLidarModeNormal = 1,      /**< Normal mode. */
  kLidarModePowerSaving = 2, /**< Power-saving mode. */
  kLidarModeStandby = 3      /**< Standby mode. */
} LidarMode;
```
而不是让激光雷达一直处于 `kLidarModeNormal`状态。

Livox-SDK, 2020-06-02:
ros driver 在运行的时候目前不能接收指令来动态切换激光雷达的运行状态；
ros driver 关闭后 lidar 会自动停止数据采样（激光器不会发光），并不是一直在采集数据

xielinjiang, 2020-06-02:
我在Livox SDK中找到了设置激光雷达Mode的函数接口，但是不知道如何调用，函数接口如下：
```
livox_status LidarSetMode(uint8_t handle, LidarMode mode, CommonCommandCallback cb, void *client_data) {
  if (device_manager().device_mode() != kDeviceModeLidar) {
    return kStatusNotSupported;
  }
  uint8_t req = static_cast<uint8_t>(mode);
  livox_status result = command_handler().SendCommand(handle,
                                                      kCommandSetLidar,
                                                      kCommandIDLidarSetMode,
                                                      &req,
                                                      sizeof(req),
                                                      MakeCommandCallback<uint8_t>(cb, client_data));
  return result;
}
```

此外，如您上述，“ros driver 关闭后 lidar 会自动停止数据采样（激光器不会发光），并不是一直在采数据”。但我在测试的过程中发现，当我停止ros驱动后，激光器确实没有发光，但激光雷达里面的电机并没有停止工作（我想做的是让里面的电机也停止工作）。

Livox-SDK, 2020-06-03:
抱歉，ros driver目前还没有更好的解决这个问题的方式，除非你直接使用Livox-SDK调用模式设置的接口来切换lidar 模式

xielinjiang, 2020-06-03:
好的，谢谢

closed 2020-06-03 by xielinjiang

tanxi-1, 2024-06-20:
您好  请问解决上诉的问题了吗？？Livox SDK中找到了设置激光雷达Mode的函数接口如何调用？？

85256638, 2025-03-01:
@xielinjiang @tanxi-1 大佬们，请问你们有进展吗？能在ROS Driver里面直接切换Lidar 模式吗？如果直接用SDK的话，那就不能同时用ROS driver了，因为激光雷达只能同时和一个节点/主机 通讯。