Livox-SDK/livox_ros_driver #32: Question about ros lvx_to_bag

hz3014, 2020-06-11:
Hi, we have a windows sdk collected lvx file contains 8 horizon lidar. Now i am trying to convert it to rosbag. While the conversion was successful, I did not see any data published on /livox/lidar topic. On the other hand, I downloaded a lvx file from your website, in this case there were data published on /livox/lidar.

Question is:
 is it possible to convert windows collected lvx file to rosbag through livox_ros_driver? if yes, what is the key element that i need to take care?

xmba15, 2020-06-13:
this livox ros sdk utilizes pure livox SDK to read lvx file, collect the packet data to form point cloud data; then finally use ROS as interface to publish /livox/lidar message. So if your collected lvx file on windows can be read by livox viewer, I think it can be normally be read by this ROS sdk itself, using the construction written in README

hz3014, 2020-06-13:
> this livox ros sdk utilizes pure livox SDK to read lvx file, collect the packet data to form point cloud data; then finally use ROS as interface to publish /livox/lidar message. So if your collected lvx file on windows can be read by livox viewer, I think it can be normally read by this ROS sdk itself, using the construction written in README

Do i need to set up broadcast_code for lvx_to_bag conversion?

Livox-SDK, 2021-04-25:
Hi hz2014,

For issue-not able to convert Horizon Point Cloud Data 2, Based on your description, I am highly suspecting that you either not typing the full file path or the lvx file name error.
Please try typing the full path or changing the lvx file name to an English name and try again

closed 2021-06-16 by Livox-SDK

TheSeanParker, 2024-05-07:
> Hi, we have a windows sdk collected lvx file contains 8 horizon lidar. Now i am trying to convert it to rosbag. While the conversion was successful, I did not see any data published on /livox/lidar topic. On the other hand, I downloaded a lvx file from your website, in this case there were data published on /livox/lidar.
> 
> Question is: is it possible to convert windows collected lvx file to rosbag through livox_ros_driver? if yes, what is the key element that i need to take care? to change the config of hub.json?

https://zhuanlan.zhihu.com/p/696431746