Livox-SDK/livox_ros_driver #33: Question about ros lvx_to_bag

onionysy, 2020-06-13:
Hello, thanks to give us the livox_ros_driver sdk. I have use this convert lvx to bag. However, there are many packet loss in the switching process. The .lvx file contains about 2 minutes of radar data, which is converted to a .bag file with only a few seconds of data.  I want to know  if  I have missed some configuration. Thx to your answer.
[image]

Livox-SDK, 2020-06-13:
Hi, onionysy, no additional configuration is required.
It looks like the timestamp of poincloud packet is wrong.
How is your pc connected with LiDAR,could you describe it ?

onionysy, 2020-06-13:
Thx for your reply. We don not connected the lidar directly. The third party record and save the data as .lvx profile and  provide it to us. And then, we try to convert .lvx to .bag file on our PC. In the switching process, the problem arises. Can you give me some advice?

EulerWong, 2020-06-15:
> Hi, onionysy, no additional configuration is required.
> It looks like the timestamp of poincloud packet is wrong.
> How is your pc connected with LiDAR,could you describe it ?

i met this problem too.
I can conver lvx file to .bag file using lvx_to_rosbag.launch. but when i run lvx_to_rosbag_rviz.launch , i met this problem too.

[ INFO] [1592203742.783323076]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783339606]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783356187]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783382875]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783401321]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783424862]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783442713]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783463636]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783484408]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783499639]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783529975]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783548993]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783566417]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783588884]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783615026]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.783634930]: Lidar[9] packet loss, interval is 10000000us
[ INFO] [1592203742.865841938]: Press [Ctrl+C] to exit!

Lvx to rosbag convert complete and exit!
[ INFO] [1592203743.178358236]: Subscribing to /clicked_point
[ INFO] [1592203743.179893851]: Subscribing to /livox/lidar
[ INFO] [1592203743.180886429]: Subscribing to /move_base_simple/goal
[ INFO] [1592203743.181926516]: Subscribing to /initialpose

and the converted file us 4.1KB size.

Livox-SDK, 2020-06-16:
Hi,onionysy, you can send a lvx file to dev@livoxtech.com, we will have a test.

closed 2020-08-12 by onionysy