Livox-SDK/livox_ros_driver #36: Mid-100设备无法订阅三个Mid-40的一帧

EulerWong, 2020-06-24:
Livox Team， 你们好，我们的设备型号是Mid-100， 操作系统Ubuntu 18.04, 我们连接了设备，运行 

roslaunch livox_ros_driver livox_lidar_msg.launch

进行/livox/lidar话题发布，我们订阅这个话题（/livox/lidar）的时候，发现在订阅回调函数里每次被调用的时候接收到的只是其中一个Mid-40的点云数据，观察发现，好像是这三个Mid-40分别按依次发布的。不能一次性在调用订阅的回调函数的直接返回三个Mid-40的一起的一帧点云数据吗？目前每次回调函数只能接收到其中某一个Mid-40返回的点云。怎样才能让回调函数接收的是Mid-100（三个Mid-40）的一帧点云数据呢？

我们的配置：
publish_freq：10
multi_topic：0
xfer_format：2

Livox-SDK, 2020-06-27:
你好，EulerWong，Mid-100设备在逻辑上相当于三个独立的Mid-40，数据上也是独立的；在livox custom msg 格式的点云包中，每个点都是有时间戳的，应用层可以通过时间戳来对齐来自不同LiDAR的点云信息.

EulerWong, 2020-06-28:
> 你好，EulerWong，Mid-100设备在逻辑上相当于三个独立的Mid-40，数据上也是独立的；在livox custom msg 格式的点云包中，每个点都是有时间戳的，应用层可以通过时间戳来对齐来自不同LiDAR的点云信息.

就是说我要在回调函数中：
void Callback(const sensor_msgs::PointCloud2ConstPtr &cloud_msg)
{
   
}
我根据时间戳，自己获取特定时长的一帧点云数据是吗？比如不管我ROS中点云的发布频率是多少，如果我要获取0.7秒的点云数据，我就自己在回调函数中自己根据时间戳计数是吗？
你们有示例代码吗？或者你们可以贴个示例吗，我现在都获取不到时间戳。

Livox-SDK, 2020-06-28:
[custom msg 点云包数据格式](https://github.com/Livox-SDK/livox_ros_driver/blob/master/README_CN.md#43-%E8%A7%88%E6%B2%83-ros-%E9%A9%B1%E5%8A%A8%E7%A8%8B%E5%BA%8F%E7%82%B9%E4%BA%91%E6%95%B0%E6%8D%AE%E8%AF%A6%E7%BB%86%E8%AF%B4%E6%98%8E)

EulerWong, 2020-06-28:
> [custom msg 点云包数据格式](https://github.com/Livox-SDK/livox_ros_driver/blob/master/README_CN.md#43-%E8%A7%88%E6%B2%83-ros-%E9%A9%B1%E5%8A%A8%E7%A8%8B%E5%BA%8F%E7%82%B9%E4%BA%91%E6%95%B0%E6%8D%AE%E8%AF%A6%E7%BB%86%E8%AF%B4%E6%98%8E)

我把点云包数据格式改成custom msg类型，即 xfer_format改为 1 后， 
void Callback()
{

}
那回调函数的形参是什么样的呢，不知道是什么类型

closed 2020-09-07 by Livox-SDK