Livox-SDK/livox_ros_driver #37: Error in lvx_to_rosbag.launch

suzuki3366, 2020-06-29:
I am trying to convert lvx files from LivoxViewer to rosbag files.
I'm using LivoxHorizon.
I use the command (`roslaunch livox_ros_driver lvx_to_rosbag. launch lvx_file_path:="xxxxx"`)
Errors :
```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:37899/

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: test.lvx
 * /cmdline_str: 100000000000000
 * /data_src: 2
 * /multi_topic: 0
 * /output_data_type: 1
 * /publish_freq: 10.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /xfer_format: 1

NODES
  /
    livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)

auto-starting new master
process[master]: started with pid [17649]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3b98411e-b9d6-11ea-91e3-000c2904aa6a
process[rosout-1]: started with pid [17662]
started core service [/rosout]
process[livox_lidar_publisher-2]: started with pid [17671]
[ INFO] [1593414037.262532016]: Livox Ros Driver Version: 2.0.1
[ INFO] [1593414037.277491507]: Data Source is lvx file.
[ INFO] [1593414037.278872809]: Create bag file :test.bag!
Open test.lvx file fail[1]!
[ERROR] [1593414037.278990743]: Init lds lvx file fail!
^C[livox_lidar_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Thank you.

Livox-SDK, 2020-06-29:
lvx_file_path:="xxxxx" , repalce the "xxxxx" with the full path of the lvx file please！

suzuki3366, 2020-06-30:
I have successfully converted from *.lvx to *.bag.
Thank you!

closed 2020-06-30 by suzuki3366