Livox-SDK/livox_ros_driver #38: rosbag does not contain original timestamps

codegrafix, 2020-06-29:
Hi,

I just noticed that a created rosbag from a lvx file does not contain the correct timestamp (always starts from time since epoch)

Can you please set the correct timestamp from the original lvx file or just let me know, if I am doing something wrong?
Thanks in advance.

Livox-SDK, 2020-07-03:
Hi, codegrafix
   The timestamp of topic in rosbag comes from the original time of pointcloud data.
    If you sync the lidar with gps time(utc), you can covert the timestamp back to UTC.

codegrafix, 2020-07-09:
Due to the missing time I now convert the rosbag message stamps to the correct timestamps manually.

Could you please add an option to the converter that uses e.g. the timestamp of the livox file?

closed 2020-09-07 by Livox-SDK

YangSiri, 2021-04-01:
Hi, @codegrafix 

What do you mean by manully correct it? Now i also need the ros timestamp of the msg. How to calculate it from the original stamp?

Looking forward to your help :)

codegrafix, 2021-04-03:
Hi @YangSiri,

In order to set the "correct" timestamps inside the converted ros file, I am parsing the timestamp from the $filename.lvx and combine it with the *header.stamp by rewriting each rosbag as follows: http://wiki.ros.org/rosbag/Cookbook

Cheers

YangSiri, 2021-04-04:
Thanks, @codegrafix 

I'm still a little confused by the __*header.stamp__ you mentioned. Actually i just want to know how to convert original timestamp in /livox/lidar msg to ros timestamp which is written in other lidar ros drivers such as Velodyne-16.

Looking forward to your help :)