Livox-SDK/livox_ros_driver #39: catkin_make error

pch9520, 2020-07-01:
```
In file included from /home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:25:0:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.h:92:7: error: ‘DeviceParameterResponse’ has not been declared
       DeviceParameterResponse *response, void *clent_data);
       ^
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp: In static member function ‘static void livox_ros::LdsLidar::OnDeviceChange(const DeviceInfo*, DeviceEvent)’:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:369:49: error: ‘LidarEnableHighSensitivity’ was not declared in this scope
                                      g_lds_ldiar);
                                                 ^
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:373:50: error: ‘LidarDisableHighSensitivity’ was not declared in this scope
                                       g_lds_ldiar);
                                                  ^
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp: At global scope:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:547:5: error: ‘DeviceParameterResponse’ has not been declared
     DeviceParameterResponse *response, void *clent_data) {
     ^
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp: In static member function ‘static void livox_ros::LdsLidar::SetHighSensitivityCb(livox_status, uint8_t, int*, void*)’:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:567:45: error: ‘LidarEnableHighSensitivity’ was not declared in this scope
                                  g_lds_ldiar);
                                             ^
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.cpp:570:46: error: ‘LidarDisableHighSensitivity’ was not declared in this scope
                                   g_lds_ldiar);
                                              ^
[ 84%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/timesync/user_uart/user_uart.cpp.o
livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:158: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lds_lidar.cpp.o' failed
make[2]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lds_lidar.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/livox_ros_driver.cpp:32:0:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.h:92:7: error: ‘DeviceParameterResponse’ has not been declared
       DeviceParameterResponse *response, void *clent_data);
       ^
livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:230: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/livox_ros_driver.cpp.o' failed
make[2]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/livox_ros_driver.cpp.o] Error 1
In file included from /home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lddc.cpp:37:0:
/home/dfg/Livox/ws_livox/src/livox_ros_driver/livox_ros_driver/lds_lidar.h:92:7: error: ‘DeviceParameterResponse’ has not been declared
       DeviceParameterResponse *response, void *clent_data);
       ^
livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:206: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o' failed
make[2]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o] Error 1
CMakeFiles/Makefile2:3403: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all' failed
make[1]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
hello, when I use catkin_make to make this package, I meet this problem, do you know how to solve it, thank you!

pch9520, 2020-07-01:
I have solved my problem!
The solution is to update Livox-SDK from 2.0.0 to 2.1.1

Caicaiyoko, 2020-07-22:
I still get this problem and my livox-SDK is 2.1.1.
I'm very comfused now.

/usr/bin/ld: warning: libboost_chrono.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_chrono.so.1.70.0
/usr/bin/ld: warning: libboost_regex.so.1.65.1, needed by /opt/ros/melodic/lib/librosconsole.so, may conflict with libboost_regex.so.1.70.0
/usr/bin/ld: warning: libboost_program_options.so.1.65.1, needed by /opt/ros/melodic/lib/librospack.so, may conflict with libboost_program_options.so.1.70.0
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libapr-1.so: undefined reference to `uuid_generate@UUID_1.0'
collect2: error: ld returned 1 exit status
livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:504: recipe for target '/home/jmrd/3D/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node' failed
make[2]: *** [/home/jmrd/3D/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node] Error 1
CMakeFiles/Makefile2:908: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all' failed
make[1]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j48 -l48" failed

pch9520, 2020-07-22:
Hi, @Caicaiyoko 
You can use "locate libboost_program_options.so" to find which version you installed for libboost_program_options.so
I think this is version problem.
This is my result
```
/opt/nvidia/nsight-systems/2019.6.2/host-linux-x64/libboost_program_options.so.1.70.0
/usr/lib/x86_64-linux-gnu/libboost_program_options.so
/usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.58.0
/usr/local/cuda-10.1/NsightSystems-2018.3/Host-x86_64/libboost_program_options.so.1.63.0
```

Caicaiyoko, 2020-07-23:
> Hi, @Caicaiyoko
> You can use "locate libboost_program_options.so" to find which version you installed for libboost_program_options.so
> I think this is version problem.
> This is my result
>
> ``` [...]

Thank you for your asnwer.
The reason is I have installed pclpy, which conflicted with the searthing for pcl.

closed 2020-09-07 by Livox-SDK