Livox-SDK/livox_ros_driver #42: [ERROR] Init lds lvx file fail!

cai-jiayou, 2020-07-03:
Hi I want to convert lvx file to .bag. 
Erro info:Open livox_test.lvx file fail[1]!
[ERROR] Init lds lvx file fail!

Livox-SDK, 2020-07-03:
Hi,cai-jiayou
    Input the full path of lvx file in the command line when launch ros driver please! 
   For example, if your lvx file  is under /home/livox/livox_test.lvx, the command is:
   `roslaunch livox_ros_driver lvx_to_rosbag.launch lvx_file_path:="/home/livox/livox_test.lvx"`

cai-jiayou, 2020-07-03:
Dear all,
Thanks for your reply.
I have did as your advice. Another error shown [livox_lidar_publisher-1] has died! Please check the attachment picture. How should I do?

Best regards,
Cai

cdq8906@126.com
 
From: livox
Date: 2020-07-03 14:14
To: Livox-SDK/livox_ros_driver
CC: cai-jiayou; Author
Subject: Re: [Livox-SDK/livox_ros_driver] [ERROR] Init lds lvx file fail! (#42)
Hi,cai-jiayou
Input the full path of lvx file in the command line when launch ros driver please!
For example, if your lvx file is under /home/livox/livox_test.lvx, the command is:
roslaunch livox_ros_driver lvx_to_rosbag.launch lvx_file_path:="/home/livox/livox_test.lvx"

Livox-SDK, 2020-07-03:
Hi,cai-jiayou,
   Is your livox_ros_driver the latest version(2.5.0)?
   If your device is Horizon, you can change the imu rate to "0".

cai-jiayou, 2020-07-03:
Yes,the Livox_ros_driver_version is 2.5.0
The .lvx file I wanted to process is captured by Mid-100,download from your official web.

cdq8906@126.com
 
From: livox
Date: 2020-07-03 16:09
To: Livox-SDK/livox_ros_driver
CC: cai-jiayou; Author
Subject: Re: [Livox-SDK/livox_ros_driver] [ERROR] Init lds lvx file fail! (#42)
Hi,cai-jiayou,
Is your livox_ros_driver the latest version(2.5.0)?
If your device is Horizon, you can change the imu rate to "0".

hashhashu, 2020-07-03:
hi,i also met such problem when i tried to convert the  lvx file to rosbag. 
my error information is" REQUIRED process [livox_lidar_publisher-2] has died!
and i also use the latest version.

hashhashu, 2020-07-04:
i've solved the problem of mine.
 i change the boost version 1.7.1 to version 1.6.5 and it works.
and i hope it can be helpful.

TheSeanParker, 2024-05-07:
> i've solved the problem of mine. i change the boost version 1.7.1 to version 1.6.5 and it works. and i hope it can be helpful.

https://zhuanlan.zhihu.com/p/696431746