Livox-SDK/livox_ros_driver #43: Simple tf causes "Message removed because it is too old"

satoshimiya, 2020-07-07:
Hi, I have added very simple static tf publisher in "livox_lidar_rviz.launch" but point cloud 2 is not published with the error saying "Message removed because it is too old".
`<node pkg="tf" type="static_transform_publisher" name="cab_livox_tf" args="1 0 1 0 0.785398 0 cab_frame livox_frame 100"/>`
The point cloud2 was displayed without this single line. Only this single line is causing error somehow. I have checked tf tree but it looks fine. As this is not rosbag, typical answer to this error such as  "set /use_simu_time true" does not work. It would be highly appreciated if someone can provide a solution to this issue.

Livox-SDK, 2020-09-07:
Hi,satoshimiya,
   You can first synchcronize the lidar time with gps, the time of lidar is the absolute time after turning on.

RakVish1342, 2021-04-29:
@satoshimiya, I had the same issue with the Livox and was wondering the same thing; Unlike while rosbag files which can publish the simulation clock using the `--clock` flag, the Livox does not seem to publish into /clock. So how can this issue of livox old messages be resolved?

@Livox-SDK, if I do not have any other sensors along with the Livox (not even a GPS sensor), how can I synchronize Livox messages with the current ROS time?

I used a hacky work around to resolve the issue. I created a node that subscribes to `/livox/lidar` and `/livox/imu`. Within the callback functions of each of the topics, I modify the incoming message's msg.header.stamp with the `ros::Time::now()` or `rospy.Time.now()` and publish it back into a different topic. I'm not sure if this follows ROS best practices but it seems to work for the isolated sensor of just the Livox.

gongyue666, 2022-02-28:
@RakVish1342 is it working for your method? I face the same problem. Does Livox can not show pointclouds in RViz with a static_tf_publisher or it can not work in our codes too?

RakVishThor, 2022-03-01:
@gongyue666, I do not use the livox lidar any more. However, the technique that I had mentioned in the workaround should resolve your rviz visualization issue. In my workaround I write a subsrciber that subscribes to the ```/livox/lidar``` and ```/livox/imu``` topics. The subscriber add timestamps manually to the ```Header``` of the messages using ```ros::Time::now()```. Then I re-publish those messages (now containing the manually added time stamps) to a  new topic like ```/livox/lidar_new``` and ```/livox/imu_new```. I record and visualize only the ```_new``` topics. While playing back the rosbag file remember to use the ```--clock``` flag to use the rosbag's ```sim_time``` rather than the onboard/computer time.

RakVish1342, 2022-03-01:
@gongyue666, I do not use the livox lidar any more. However, the technique that I had mentioned in the workaround should resolve your rviz visualization issue. In my workaround I write a subsrciber that subscribes to the ```/livox/lidar``` and ```/livox/imu``` topics. The subscriber add timestamps manually to the ```Header``` of the messages using ```ros::Time::now()```. Then I re-publish those messages (now containing the manually added time stamps) to a new topic like ```/livox/lidar_new``` and ```/livox/imu_new```. I record and visualize only the ```_new``` topics. While playing back the rosbag file remember to use the ```--clock``` flag to use the rosbag's ```sim_time``` rather than the onboard/computer time.

gongyue666, 2022-03-02:
@RakVish1342 Thx very much～