Livox-SDK/livox_ros_driver #46: How does the return_mode parameter works in the livox_lidar_config.json file?

Zlank, 2020-07-23:
Hello,

I'm trying to see the second return points in the point cloud that Livox mid 40 gives to me. When I use the standard firmware (not the multirreturn firmware), and set return_mode to 2, I can see via RVIZ and rostopic echo that there aren't second return points in the point cloud.

Then I tested the opposite, dual_return set to 0 and use the multirreturn special firmware. Now I can see the second return points. But it shouldn't work like that, I think.

I'm only using the LiDAR and a laptop to do this kind of experiments.

The question is, how does the return_mode works? Have I missed something?

Livox-SDK, 2020-07-27:
Zlank，
    The standard firmware of Mid40 does not support dual return mode，if livox ros driver recognizes the current LiDAR deivce is mid40, the return mode will not be set .

Zlank, 2020-07-27:
Ok, thanks so much for the fast response.

Best regards.

closed 2020-08-06 by Zlank