Livox-SDK/livox_ros_driver #54: No tf data. Actual error: Fixed Frame [livox_frame] does not exist

SeymurD, 2020-09-09:
Hello, 

I am having trouble visualising the point cloud using the following ROS driver on the Jetson Tx2. The problem seems to be a result of the following message:

> No tf data. Actual error: Fixed Frame [livox_frame] does not exist.

The following is the roslaunch console message which seems to show no errors:

>SUMMARY

> ========
> 
> PARAMETERS
>  * /cmdline_file_path: livox_test.lvx
>  * /cmdline_str: 100000000000000
>  * /data_src: 0
>  * /frame_id: livox_frame
>  * /multi_topic: 0
>  * /output_data_type: 0
>  * /publish_freq: 10.0
>  * /rosdistro: melodic
>  * /rosversion: 1.14.9
>  * /user_config_path: /home/seymur/Docu...
>  * /xfer_format: 0
> 
> NODES
>   /
>     livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)
>     rviz (rviz/rviz)
> 
> auto-starting new master
> process[master]: started with pid [15552]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 46dca2ea-f252-11ea-a8da-00044bf848ae
> process[rosout-1]: started with pid [15563]
> started core service [/rosout]
> process[livox_lidar_publisher-2]: started with pid [15570]
> [ INFO] [1599624580.401020910]: Livox Ros Driver Version: 2.5.0
> [ INFO] [1599624580.413450077]: Data Source is raw lidar.
> [ INFO] [1599624580.416246372]: Config file : /home/seymur/Documents/Projects/Livox_ROS/ws_livox/src/livox_ros_driver/config/livox_lidar_config.json
> Commandline input bd:100000000000000
> Invalid bd:100000000000000!
> Livox SDK version 2.1.1
> broadcast code[1HDDH1200104011] : 1 1 0 0 1 0
> Add Raw user config : 1HDDH1200104011 
> Disable timesync
> Disable auto connect mode!
> List all broadcast code in whiltelist:
> 1HDDH1200104011
> Livox-SDK init success!
> [ INFO] [1599624580.417805909]: Init lds lidar success!
> process[rviz-3]: started with pid [15572]
> 

I will also include the config file params for reference:

> {
>     "lidar_config": [
>         {
>             "broadcast_code": "1HDDH1200104011",
>             "enable_connect": true,
>             "enable_fan": true,
>             "return_mode": 0,
>             "coordinate": 0,
>             "imu_rate": 1,
>             "extrinsic_parameter_source": 0,
>             "enable_high_sensitivity": false
>         }
>     ],
> 	
>     "timesync_config": {
>         "enable_timesync": false,
>         "device_name": "/dev/ttyUSB0",
>         "comm_device_type": 0,
>         "baudrate_index": 2,
>         "parity_index": 0     
>     }
> }

I did some research through the Google Conversation forums and just browsing some of the potential things that can cause this problem but I am not having any luck.

I will also include the Static IP config file and `ifconfig` output for `eth0`:

/etc/network/interfaces

> # This file describes the network interfaces available on your system
> # and how to activate them. For more information, see interfaces(5).
> 
> # The loopback network interface
> auto lo
> iface lo inet loopback
> 
> # The primary network interface
> allow-hotplug eth0
> iface eth0 inet static
>         address 192.168.1.101
>         netmask 255.255.255.0
>         gateway 192.168.1.1

ifconfig:

> eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
>         inet6 fe80::12f9:4e8:1744:898  prefixlen 64  scopeid 0x20<link>
>         ether 00:04:4b:f8:48:b0  txqueuelen 1000  (Ethernet)
>         RX packets 5752  bytes 356480 (356.4 KB)
>         RX errors 0  dropped 0  overruns 0  frame 0
>         TX packets 1856  bytes 322436 (322.4 KB)
>         TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
>         device interrupt 41

I would really appreciate some assistance on this issue. Thanks

Livox-SDK, 2020-09-11:
Hi,SeymurD
   It seems that the TX2 can't receive any ethernet packets from our LiDAR,you can check the TX2's firewall settings.
And you can use tcpdump to dump the ethernet packets.

mick-h, 2021-02-19:
I'm getting the same error.  I'm running the demo on a linux machine.  When I run `roslaunch livox_ros_driver livox_lidar_rviz.launch` I get the following output:

> ... logging to /home/mick/.ros/log/62845878-727f-11eb-82d7-000c293af779/roslaunch-ubuntu-45064.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ubuntu:33547/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /cmdline_file_path: livox_test.lvx
>  * /cmdline_str: 100000000000000
>  * /data_src: 0
>  * /enable_imu_bag: True
>  * /enable_lidar_bag: True
>  * /frame_id: livox_frame
>  * /multi_topic: 0
>  * /output_data_type: 0
>  * /publish_freq: 10.0
>  * /rosdistro: melodic
>  * /rosversion: 1.14.10
>  * /user_config_path: /home/mick/ws_liv...
>  * /xfer_format: 0
> 
> NODES
>   /
>     livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)
>     rviz (rviz/rviz)
> 
> auto-starting new master
> process[master]: started with pid [45074]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 62845878-727f-11eb-82d7-000c293af779
> process[rosout-1]: started with pid [45085]
> started core service [/rosout]
> process[livox_lidar_publisher-2]: started with pid [45092]
> process[rviz-3]: started with pid [45093]
> [ INFO] [1613717701.047655623]: Livox Ros Driver Version: 2.6.0
> [ INFO] [1613717701.050872622]: Data Source is raw lidar.
> [ INFO] [1613717701.051231901]: Config file : /home/mick/ws_livox/src/livox_ros_driver/config/livox_lidar_config.json
> Commandline input bd:100000000000000
> Invalid bd:100000000000000!
> Livox SDK version 2.3.0
> broadcast code[3WEDH7600102721] : 1 1 0 0 0 0
> Add Raw user config : 3WEDH7600102721 
> Disable timesync
> Disable auto connect mode!
> List all broadcast code in whiltelist:
> 3WEDH7600102721
> Livox-SDK init success!
> [ INFO] [1613717701.052039384]: Init lds lidar success!
> [2021-02-18 22:55:01.226] [console] [info]  Broadcast broadcast code: 3WEDH7600102721  [device_discovery.cpp] [OnBroadcast] [150]
> [2021-02-18 22:55:01.226] [console] [info] LocalIP: 192.168.1.205  [device_discovery.cpp] [OnBroadcast] [201]
> [2021-02-18 22:55:01.226] [console] [info] DeviceIP: 192.168.1.172  [device_discovery.cpp] [OnBroadcast] [202]
> [2021-02-18 22:55:01.226] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
> [2021-02-18 22:55:01.226] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
> [2021-02-18 22:55:01.250] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
> [2021-02-18 22:55:01.250] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
> [2021-02-18 22:55:01.250] [console] [info] Broadcast Code: 3WEDH7600102721  [device_discovery.cpp] [OnData] [101]
> [2021-02-18 22:55:01.250] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [102]
> [2021-02-18 22:55:01.250] [console] [info] IP: 192.168.1.172  [device_discovery.cpp] [OnData] [103]
> [2021-02-18 22:55:01.250] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
> [2021-02-18 22:55:01.250] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
> [2021-02-18 22:55:01.250] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [243]
> [2021-02-18 22:55:01.274] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [118]
> Lidar[3WEDH7600102721] status_code[0] working state[5] feature[0]
> [2021-02-18 22:55:01.274] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
> [2021-02-18 22:55:01.298] [console] [info]  Update State to 1, device connect true  [device_manager.cpp] [UpdateDeviceState] [257]
> Lidar[3WEDH7600102721] status_code[0] working state[1] feature[0]
> [2021-02-18 22:55:01.298] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
> firmware version: 6.10.0.0
> [2021-02-18 22:55:01.298] [console] [info]  Send Command: Set 0 Id 5 Seq 6  [command_channel.cpp] [Send] [243]
> [2021-02-18 22:55:01.298] [console] [info]  Send Command: Set 1 Id 6 Seq 7  [command_channel.cpp] [Send] [243]
> [2021-02-18 22:55:01.298] [console] [info]  Send Command: Set 1 Id 8 Seq 8  [command_channel.cpp] [Send] [243]
> [2021-02-18 22:55:01.322] [console] [info]  Recieve Ack: Set 0 Id 5 Seq 6  [command_handler.cpp] [OnCommand] [118]
> Set coordinate success!
> [2021-02-18 22:55:01.322] [console] [info]  Recieve Ack: Set 1 Id 6 Seq 7  [command_handler.cpp] [OnCommand] [118]
> Set return mode success!
> [2021-02-18 22:55:01.322] [console] [info]  Recieve Ack: Set 1 Id 8 Seq 8  [command_handler.cpp] [OnCommand] [118]
> Set imu rate success!
> [2021-02-18 22:55:01.322] [console] [info]  Send Command: Set 0 Id 4 Seq 9  [command_channel.cpp] [Send] [243]
> Lidar[0][3WEDH7600102721] DataType[2] PacketInterval[400032] PublishPackets[250]
> Lidar[0][3WEDH7600102721] storage queue size : 500 512
> [2021-02-18 22:55:01.347] [console] [info]  Recieve Ack: Set 0 Id 4 Seq 9  [command_handler.cpp] [OnCommand] [118]
> Lidar start sample success
> [ INFO] [1613717701.424086009]: Support only one topic.
> [ INFO] [1613717701.425031036]: livox/lidar publish use PointCloud2 format, set ROS publisher queue size 256

I then see the mentioned error in RViz, and no data is plotted on the grid, but also can see that messages are being received and that max intensity varies as I move the Horizon around.  So the data seems to be being received by the computer, but still the same error and nothing displayed.

Any other ideas as to what might cause this?

pch9520, 2021-03-22:
> I'm getting the same error. I'm running the demo on a linux machine. When I run `roslaunch livox_ros_driver livox_lidar_rviz.launch` I get the following output:
>
>
> I then see the mentioned error in [...]

I meet the same problem !

nes9966, 2022-09-14:
Hello.  I am running ubuntu 18.04 on a virtual box.
I have updated the the livox_ros_driver , but still face the same issue.  Has anyone found an answer to this?