Livox-SDK/livox_ros_driver #55: Command Port and Data Port are constantly changing

lyh2833495, 2020-09-15:
After starting the livox_ros_driver, console prints the successful information of connecting Lidar, and prints the correct BD code. However, the Command Port and Data Port are always changing, and there is no topic published by the Lidar, at least I cant find it by "rostopic list".

What should I do in this situation?

Thanks in advance.

Livox-SDK, 2020-09-16:
Hi,lyh2833495
    What are the details printed on the console, can you list it here?

lyh2833495, 2020-09-20:
like this:
===============================
Broadcast broadcast code:1HDDH1200105851 [device_discovery.cpp][OnBroadcast][162]
LocalIP:192.168.1.51 [device_discovery.cpp][OnBroadcast][225]
Command Port:55518 [device_discovery.cpp][OnBroadcast][226]
Data Port:56018 [device_discovery.cpp][OnBroadcast][227]

Broadcast broadcast code:1HDDH1200105851 [device_discovery.cpp][OnBroadcast][162]
LocalIP:192.168.1.51 [device_discovery.cpp][OnBroadcast][225]
Command Port:55519 [device_discovery.cpp][OnBroadcast][226]
Data Port:56019 [device_discovery.cpp][OnBroadcast][227]

Broadcast broadcast code:1HDDH1200105851 [device_discovery.cpp][OnBroadcast][162]
LocalIP:192.168.1.51 [device_discovery.cpp][OnBroadcast][225]
Command Port:55520 [device_discovery.cpp][OnBroadcast][226]
Data Port:56020 [device_discovery.cpp][OnBroadcast][227]

Broadcast broadcast code:1HDDH1200105851 [device_discovery.cpp][OnBroadcast][162]
LocalIP:192.168.1.51 [device_discovery.cpp][OnBroadcast][225]
Command Port:55521 [device_discovery.cpp][OnBroadcast][226]
Data Port:56021 [device_discovery.cpp][OnBroadcast][227]

..........
===========================================
Command Port and Data Port are constantly changing

lyh2833495, 2020-09-26:
We solved this problem. The cause of the problem is the network cable connection.

Thank you!

kisdia, 2020-11-30:
Can you explain what do you mean by "problem is the network cable connection" ?
We are having the same problem.