Livox-SDK/livox_ros_driver #56: It is converted to only 4 kb regardless of the original capacity.

dmmp360view, 2020-09-16:
Hello
Do I have to give an option to convert lvx file to bag file so that it can be converted normally?
If you check the information in the bag file after conversion, the version and size will be shown below. Regardless of the original size.(4.0kb)
Is there any way to solve this problem?

command
# roslaunch livox_ros_driver lvx_to_rosbag_rviz.launch  lvx_file_path:="/home/livox/test.lvx"
#rosbag info /home/livox/test.bag

result
path : /home/livox/test.bag
version : 2.0
size : 4.0kb

Livox-SDK, 2020-09-16:
Replace ("/home/livox/test.lvx") with your livox file path please, thanks!

dmmp360view, 2020-09-16:
Of course I did. I'm not asking you that.!!!!!  
It's an example. I've already converted a lot of files with that command. No errors.
When converting an lvx file to a bag file, I am asking you how to use the parameter.
ex) roslaunch livox_ros_driver lvx_to_rosbag_rviz.launch lvx_file_path:="/home/livox/test.lvx"
If I don't use the parameter, does it not convert properly? What's the default setting?

Livox-SDK, 2021-04-21:
Hi dmmp360view,
Hmm, let me explain this comment first 
roslaunch livox_ros_driver lvx_to_rosbag_rviz.launch lvx_file_path:="/home/livox/test.lvx"
**run-livox ros dirver**            **convert the test.lvx under directory "/home/livox/ "to rosbag file** 
yes, this parameter( lvx_to_rosbag_rviz.launch lvx_file_path:="/home/livox/test.lvx") is possible to be changed. if you changed your lvx-file's name or put the lvx file under another directory
the rosbag file converted, will be put under the same directory that's the reason why when you check the detail of the rosbag file the path showed "/home/livox/test.bag"

let me make another example, if you have another lvx file need to be converted named: test_1.lvx under directory " /home/dmmp360view/" , just simply type:
roslaunch livox_ros_driver lvx_to_rosbag_rviz.launch lvx_file_path:="/home/dmmp360view/test_1.lvx"

hope you found this answer is helpful.

closed 2021-06-16 by Livox-SDK