Livox-SDK/livox_ros_driver #62: How to differentiate /livox/imu messages in case of hub configuration

radeonwu, 2020-10-27:
Would like to check with Livox folks, that in case multi horizon lidars are installed with a hub unit, since every lidar unit publish the same /livox/imu message, which is of type sensor_msgs/Imu, how to seperate these imu messages so that we know which lidar unit there are attached to? Thanks

Livox-SDK, 2020-10-28:
Hi,radeonwu,
    You can set the parameters "multi_topic" to "1", Each LiDAR imu data will be published using its own independent publisher.

radeonwu, 2020-10-30:
> 
> 
> Hi,radeonwu,
> You can set the parameters "multi_topic" to "1", Each LiDAR imu data will be published using its own independent publisher.

Thanks for the reply.
I am aware about the ros param setting of "multi_topic".   How about existing dataset that was recorded with multi horizon lidars and a single /livox/imu message inside, any possibility to seperate the imu message?

Livox-SDK, 2020-11-02:
Hi,radeonwu,
   Is the format of the “existing dataset” file .lvx or .bag?  If .lvx file, you can convert it to rosbag using multi_topic;
But if your dataset file is already a rosbag, it‘s impossible to seperate the message.

radeonwu, 2020-11-03:
thanks for your reply.
unfortunately the data was recorded in .bag format.

closed 2020-11-07 by radeonwu