Livox-SDK/livox_ros_driver #67: PointCloud format

YangSiri, 2020-11-13:
Hi, @Livox-SDK  

I'm curious about the PointXYZRTL format of livox lidar point cloud. Are all livox series support this? What does the tag and line parameter mean?

Loking forward to your help ;>

Livox-SDK, 2020-11-20:
Hi,YangSiri
MID-40/MID-100 doesn't support the tag flag, but Horizon/Tele/Mid70/Mid70Pro support it.
The Customized point cloud format is livox defined format , the description of this pointcloud format is in README.md

YangSiri, 2020-11-21:
Thanks, @Livox-SDK 

Could you please show me where i can find the understanding of _tag_ and _line_ flag? And do all livox lidars support Customized point cloud format?

Livox-SDK, 2020-11-21:
Hi,YangSiri
    you can reference the following link:
[Livox-SDK-Communication-Protocol](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol)

closed 2021-06-16 by Livox-SDK

goldqiu, 2022-10-21:
Is there only one lasers in Mid70？   I also printf the information of the ‘line’ ,but it has no output information. Could you please answer my question? Thank you.