Livox-SDK/livox_ros_driver #68: ubuntu18.04 编译livox_ros_driver报错

helloworld202, 2020-11-19:
[image]

Livox-SDK, 2020-11-20:
（1）Remove the builder folder in livox_ros_driver directory
rm -rf build
（2）rebuild 
catkin_build

kxch928925819, 2021-03-29:
[ 76%] Built target livox_ros_driver_generate_messages
[ 80%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/timesync/timesync.cpp.o
[ 84%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/timesync/user_uart/user_uart.cpp.o
[ 88%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/common/comm/comm_protocol.cpp.o
[ 92%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/common/comm/sdk_protocol.cpp.o
[ 96%] Building CXX object livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/common/comm/gps_protocol.cpp.o
[100%] Linking CXX executable /home/dji/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::CreateBagFile(std::string const&)':
lddc.cpp:(.text+0x41b): undefined reference to `rosbag::Bag::open(std::string const&, unsigned int)'
lddc.cpp:(.text+0x4ec): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::PrepareExit()':
lddc.cpp:(.text+0x677): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
lddc.cpp:(.text+0x6f0): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::GetCurrentPublisher(unsigned char)':
lddc.cpp:(.text+0x13ca): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
lddc.cpp:(.text+0x1459): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o:lddc.cpp:(.text+0x1539): more undefined references to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)' follow
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::PublishPointcloud2(livox_ros::LidarDataQueue*, unsigned int, unsigned char)':
lddc.cpp:(.text+0x2016): undefined reference to `ros::Publisher::getTopic() const'
lddc.cpp:(.text+0x239c): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::PublishPointcloudData(livox_ros::LidarDataQueue*, unsigned int, unsigned char)':
lddc.cpp:(.text+0x270e): undefined reference to `ros::Publisher::getTopic() const'
lddc.cpp:(.text+0x2ad7): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::PublishCustomPointcloud(livox_ros::LidarDataQueue*, unsigned int, unsigned char)':
lddc.cpp:(.text+0x2fcb): undefined reference to `ros::Publisher::getTopic() const'
lddc.cpp:(.text+0x312a): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `livox_ros::Lddc::PublishImuData(livox_ros::LidarDataQueue*, unsigned int, unsigned char)':
lddc.cpp:(.text+0x3574): undefined reference to `ros::Publisher::getTopic() const'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `void rosbag::Bag::writeMessageDataRecord<sensor_msgs::PointCloud2_<std::allocator<void> > >(unsigned int, ros::Time const&, sensor_msgs::PointCloud2_<std::allocator<void> > const&)':
lddc.cpp:(.text._ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_[_ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_]+0x104): undefined reference to `rosbag::Bag::toHeaderString(ros::Time const*) const'
lddc.cpp:(.text._ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_[_ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_]+0x246): undefined reference to `rosbag::Bag::writeHeader(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
lddc.cpp:(.text._ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_[_ZN6rosbag3Bag22writeMessageDataRecordIN11sensor_msgs12PointCloud2_ISaIvEEEEEvjRKN3ros4TimeERKT_]+0x280): undefined reference to `rosbag::Bag::appendHeaderToBuffer(rosbag::Buffer&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o: In function `void rosbag::Bag::writeMessageDataRecord<pcl::PointCloud<pcl::PointXYZI> >(unsigned int, ros::Time const&, pcl::PointCloud<pcl::PointXYZI> const&)':
[... cut ...]
livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/build.make:1074: recipe for target '/home/dji/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node' failed
make[2]: *** [/home/dji/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node] Error 1
CMakeFiles/Makefile2:1012: recipe for target 'livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all' failed
make[1]: *** [livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Same to me! PCL 1.91 down to PCL1.80 ,LIVOX-SDK 2.3 down to LIVOX-SDK 2.20