Livox-SDK/livox_ros_driver #71: 跑loam_horizon时catkin_make报错，找不到livox_ros_driver

AliveForU, 2020-12-20:
按照教程装的livox_ros_driver，测试已经安装成功了，但是跑loam_horizon时，却说找不到livox_ros_driver，不知道哪里出错了

这是错误代码：
 Could not find the required component 'livox_ros_driver'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "livox_ros_driver"
  with any of the following names:

    livox_ros_driverConfig.cmake
    livox_ros_driver-config.cmake

Livox-SDK, 2020-12-23:
使用如下命令，清除一下编译输出文件：
rm -rf build
catkin_make

harzn, 2021-01-18:
按这个操作后问题依旧存在，甚至在cmakelist里set了driver的地址 也还是找不到

Gold-finger99, 2021-01-21:
编译后source ~/catkin_ws/devel/setup.bah试一下

lhypds, 2021-02-17:
same issue, cannot make

hitdezhazha, 2021-03-31:
一样的问题

inntoy, 2021-07-15:
可以把ws_livox下的src文件夹全部移到～/catkin_ws/src里，然后把这个文件夹重新命名为livox_ros_drive

BUG423, 2023-04-06:
> 

您好，该方案“rm -rf build
catkin_make”没有奏效，请问是否有其他解决方案

Gold-finger99, 2023-04-06:
已收到感谢您得来信

inntoy, 2023-04-06:
您好，您的邮件我已收到。————祝您每天都有个好心情!

BUG423, 2023-04-06:
您好，我已经收到您的来信，非常感谢！

BUG423, 2023-04-06:
你好，该问题已经解决。可以参考该文章https://blog.csdn.net/weixin_39754100/article/details/114894454（侵权请联系删除）
两种方案：
1.把整个项目文件夹复制到livox_ros_driver的src文件夹下,再在根目录下分别执行source devel/setup.bash和catkin_make便顺利编译通过
2.在当前项目的根目录打开终端,在此终端下source livox_ros_driver的bash文件,如source ~/xxx/livox_ros_driver/devel/setup.bash,然后再catkin_make即可顺利编译

jumpjumpqaq, 2024-05-22:
编译完livox_ros_driver后，会生成devel空间，devel空间下有编译所需的目录，在CMakeLists.txt添加多一行set(livox_ros_driver_DIR "/home/taotao/code/vslam/livox_driver_ws/devel/share/livox_ros_driver/cmake/")，找到livox_ros_driverConfig.cmake文件的路径

Gold-finger99, 2024-05-22:
已收到感谢您得来信

BUG423, 2024-05-22:
您好，我已经收到您的来信，非常感谢！祝好！