Livox-SDK/livox_ros_driver #76: No response from tele15 for some of my cmds

Jimmymiji, 2021-01-27:
Hi,

   I am using livox tele15. When I sent the following commands to tele15:
      -Handshake Request
      -Query Device Information
      -Set Coordinate System
      -Set Return Mode
      -Set IMU Frequency
      -Query Extrinsic Parameter
   it only respond to: 
     - Handshake Request
      -Query Device Information
  I also captured the udp packets between host server and livox to confirm this. May I know if there is something I miss but specified in the Communication Portocol. 
Thank you!

Jimmymiji, 2021-01-29:
Problem solved, some ACK msgs are put in one UDP packet ....
But I think it should be specified in the communication protocol.

Livox-SDK, 2021-01-30:
We will test this case,thanks!