Livox-SDK/livox_ros_driver #77: Error in lvx_to_rosbag. REQUIRED process [livox_lidar_publisher-2] has died!

DiantaoTu, 2021-02-04:
I use the command `roslaunch livox_ros_driver lvx_to_rosbag.launch lvx_file_path:="~/Desktop/test.lvx"` to convert **test.lvx** to **test.bag**. But an error occured.
```
REQUIRED process [livox_lidar_publisher-2] has died!
process has died [pid 19754, exit code -6, cmd /home/tdt/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/tdt/.ros/log/f1cf6550-66f8-11eb-9263-e454e868005f/livox_lidar_publisher-2.log].
log file: /home/tdt/.ros/log/f1cf6550-66f8-11eb-9263-e454e868005f/livox_lidar_publisher-2*.log
Initiating shutdown!
```
OS: Ubuntu 16.04 LTS
ROS: ros-kinetic-desktop

Looking forward to your help ^_^

yongcaiwang, 2021-02-14:
I meet the same problem. Have you solved the problem?
I am using ubuntu 18.04 LTS
ROS: ros-kinetic-desktop
Looking forward to your help ^_^

The error message is: 

================================================================================REQUIRED process [livox_lidar_publisher-2] has died!
process has died [pid 9154, exit code -6, cmd /home/ycw/catkin_ws/src/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/ycw/.ros/log/78aff22c-6ecb-11eb-afc0-50eb71604b54/livox_lidar_publisher-2.log].
log file: /home/ycw/.ros/log/78aff22c-6ecb-11eb-afc0-50eb71604b54/livox_lidar_publisher-2*.log
Initiating shutdown!
================================================================================

Livox-SDK, 2021-02-20:
<arg name="imu_bag" default="false"/>
Disable covert  imu data in lvx file please.

Yuze-HE, 2021-03-05:
>  Disable covert imu data in lvx file please.

Could you please explain more? What is the "lvx file" you are referring to?

Livox-SDK, 2021-03-06:
[image]

XW-Fan, 2021-11-10:
> [image]

Hello, I have the same question, but the "imu_bag" has been set false. Is there any other suggestions?

my error message is as follow:

[ INFO] [1636526682.905539007]: Livox Ros Driver Version: 2.6.0
[ INFO] [1636526682.907994929]: Data Source is lvx file.
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: /home/livox/test.bag
================================================================================REQUIRED process [livox_lidar_publisher-2] has died!
process has died [pid 3958, exit code -6, cmd /home/wenhui/Livox-SDK/build/sample/lidar/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/wenhui/.ros/log/af44ae3c-41f1-11ec-af4d-001e101f0000/livox_lidar_publisher-2.log].
log file: /home/wenhui/.ros/log/af44ae3c-41f1-11ec-af4d-001e101f0000/livox_lidar_publisher-2*.log
Initiating shutdown!
================================================================================
[livox_lidar_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ahmadbh96, 2022-03-25:
Hello,
I am having the same problem. But i dont think it has to do with the commands and launch files.
During the installation of livox_ros_driver, the program compiles but I get this (attached)
Please help, Ive been trying to run this for a couple of months now
[image]
.

ahmadbh96, 2022-03-25:
I think this is what causes the error when trying to convert lvx to rosbag.