Livox-SDK/livox_ros_driver #78: Could each LiDAR set frame_id when livox_hub_msg.launch?

h-wata, 2021-02-18:
I've tested livox LiDAR and Hub.
I know your released [livox_automatic_calibration](https://github.com/Livox-SDK/Livox_automatic_calibration), but It couldn't estimate well in my environment.

Therefore I would like to set  the frame_id for EACH LiDARs when livox_hub_msg.launch.
Do you have a plan to do that?

Thanks your great products and attentions.

Livox-SDK, 2021-02-20:
hi,h-wata:
you can specify the frame_id by handle.

void Lddc::InitPointcloud2MsgHeader(sensor_msgs::PointCloud2& cloud) {
  cloud.header.frame_id.assign(frame_id_);
  cloud.height = 1;
  cloud.width = 0;
  cloud.fields.resize(6);
  cloud.fields[0].offset = 0;
....
}
modify it to 
void Lddc::InitPointcloud2MsgHeader(sensor_msgs::PointCloud2& cloud, uint8_t  handle) {
format the frame id string by handle here
}

h-wata, 2021-02-22:
Thanks for your reply and solutions.
I would like to confirm as follows;
You mean I should modify this lines, right?
https://github.com/Livox-SDK/livox_ros_driver/blob/1d6e869a3e7422669825da51fc788f0128f141a1/livox_ros_driver/livox_ros_driver/lddc.cpp#L132-L162
I understand this node use in single Livox Lidar bringing up.

In my case, I would like to use multi sensors with Livox Hub and to set frame_id each Lidars.
If I misunderstand, please correct.

closed 2021-03-05 by h-wata

maomao0220, 2021-12-17:
> 感谢您的回复和解决方案。 我想确认如下； 你的意思是我应该修改这一行，对吧？
> 
> https://github.com/Livox-SDK/livox_ros_driver/blob/1d6e869a3e7422669825da51fc788f0128f141a1/livox_ros_driver/livox_ros_driver/lddc.cpp#L132-L162
> 
> 
> ~我了解此节点在单个 Livox 激光雷达中的使用。~
> ~就我而言，我想在 Livox Hub 中使用多传感器并为每个激光雷达设置 frame_id。 如果我理解错了，请纠正。~
> 
> 我了解您的解决方案。 我尝试修改`Lddc::InitPointcloud2MsgHeader`.

maomao0220, 2021-12-17:
How did you solve it? Thanks for your reply

h-wata, 2021-12-21:
> How did you solve it? Thanks for your reply

I have not been solved.
I set each lidar's positons by "Extrinsic Calibration" in LivoxViewer.
I calibrated by hand because I gave up on using automatic calibration.

I'm sorry I couldn't help you...