Livox-SDK/livox_ros_driver #80: 单个mid-70雷达，运行ros驱动， 代码卡在lds_->semaphore_.Wait()

YangLiu-Leo, 2021-03-18:
环境：ubuntu18.04  在docker里装了ros kinetic+livox-SDK
使用官方软件可以读取雷达数据，见附件图片
修改完序列号，运行roslaunch livox_ros_driver livox_lidar.launch  程序卡住，经debug程序卡在了lddc.cpp lds_->semaphore_.Wait()这个地方
请问，是什么配置有问题吗？还是代码不支持mid-70雷达

[image]
[image]
[image]

Livox-SDK, 2021-03-18:
没有连接上 lidar，先在原生 ubuntu 环境测试下吧，应该是 docker 网络环境的问题

YangLiu-Leo, 2021-03-22:
谢谢，已经解决，由于我在docker里跑的程序，可能网络端口不通。需要docker run 的时候加 --net=host，如果使用了docker-compose 可以在docker_compose.yml services下一级加 network_mode: host

closed 2021-03-22 by YangLiu-Leo