Livox-SDK/livox_ros_driver #81: Livox Hub connection issue plz help..

ChicagoPARK, 2021-03-29:
Currently, I have 5 livox horizon lidars and Iivox hub.
Even I prepare all the things which is necessary for operating Livox hub (power cable).

But when I launch "livox_hub.launch", I got the message below.

Could you give me a help??

(When I connect the one lidar individually and launch "livox_lidar.launch", I can get the topic.)

OS: Linux Ubuntu 18.04 melodic

 Error code
`... logging to /home/kaai/.ros/log/66e9b5a8-906a-11eb-b4e4-305a3a77c45e/roslaunch-kaai-PC-13855.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kaai-PC:41057/

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 100000000000000
 * /data_src: 1
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /multi_topic: 1
 * /output_data_type: 0
 * /publish_freq: 10.0
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /user_config_path: /home/kaai/catkin...
 * /xfer_format: 0

NODES
  /
    livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)

ROS_MASTER_URI=http://localhost:11311

process[livox_lidar_publisher-1]: started with pid [13870]
[ INFO] [1617008692.828183573]: Livox Ros Driver Version: 2.6.0
[ INFO] [1617008692.833528817]: Data Source is hub.
[ INFO] [1617008692.834038117]: Config file : /home/kaai/catkin_ws/src/Livox-SDK/livox_ros_driver/livox_ros_driver/config/livox_hub_config.json
Commandline input bd:100000000000000
Invalid bd:100000000000000!
Livox SDK version 2.3.0
Hub[13UUG1R00400170] : 0 0 0 0 0
Lidar[0TFDG3B006H2Z11] : 0 0 0 0 0
Add lidar raw user config : 0TFDG3B006H2Z11 
Lidar[0TFDG3U99101291] : 0 0 0 0 0
Add lidar raw user config : 0TFDG3U99101291 
Lidar[1HDDG8M00100191] : 0 0 0 0 0
Add lidar raw user config : 1HDDG8M00100191 
Lidar[1PQDG8E00100321] : 0 0 0 0 0
Add lidar raw user config : 1PQDG8E00100321 
No broadcast code was added to whitelist, swith to automatic connection mode!
Livox-SDK init success!
[ INFO] [1617008692.834802578]: Init lds hub success!
`
[image]

Livox-SDK, 2021-04-01:
Hub use static ip address by default, could you check you host machine(pc) ip address?

ChicagoPARK, 2021-04-01:
I already tried static ip (according to the Livox hub manual) "192.168.1.X". And I put the number between 2 to 233...

The above message came out after trying static ip and same with auto ip as well..

Do you mean should I need to set static ip address same with my PC internet IP?

Livox-SDK, 2021-04-01:
Hub and PC both use static ip mode,  but the ip of hub and pc must be different.
It seems that no data is sent to the PC.

Livox-SDK, 2021-04-01:
Use tcpdump or wireshark to monitor your network.
Confirm whether there is hub broadcast data sent out.

ChicagoPARK, 2021-04-01:
Copy.
Thank you for your answer.
I am goint to let you know about it after trying your recommendation

ChicagoPARK, 2021-04-06:
Sir, should I need to use Gigabit Router (DHCP) to connect ??

Currently, I connect ethernet cable directly. But I do not know it is the problem or not..

Livox-SDK, 2021-04-08:
ChicagoPARK，sorry, the default IP mode of  Livox Hub is dynamic！
I made a mistake before，I'm very very sorry！
If the Livox Hub in dynamic ip mode，you must use a Gigabit Router (DHCP)!
You can also install a DHCP server in ubuntu，please refer to the following link:
https://graspingtech.com/configure-dhcp-server/

ChicagoPARK, 2021-04-08:
Thank you for your answer
So you mean, Static IP mode is impossible to use?
If that is true I am gonna buy that

Livox-SDK, 2021-04-08:
"So you mean, Static IP mode is impossible to use?"
No,you can change the IP mode of Livox Hub to static via Livox viewer!
But you must first connect to the Livox Hub.

Livox-SDK, 2021-04-08:
Livox Hub both support dynamic IP mode and static IP mode.
But the default IP mode at the factory is dynamic.

TaekHyungCho, 2021-04-26:
Hi, I also had same issue. You said " You can also install a DHCP server in ubuntu，please refer to the following link: " 
That mean if i installed DHCP server in my pc, It is not necessary to use DHCP router? 
If that is true, After installed DHCP server, just connect hub and pc directly is the right answer to get topic message from livox_hub?

Livox-SDK, 2021-05-04:
TaekHyungCho，yes,  you can connect hub and pc directly after installing DHCP server if the hub is in dynamic IP mode.

ChicagoPARK, 2021-05-07:
Thank you for keep watching our problem sir.
We still struggle with that problem. I am gonna attach all the pictures to describe our situation.

**[First Picture : We've used five LiDAR]**
[image]

**[Second Picture : How we connect on Hub]**
[image]

**[Third Picture : The DHCP which you ask us to utilize. And we connect Hub and Pc with this]**
[image]

[Fourth Picture : IPv4, IPv6 information. As you can see we set the mode as a DHCP]
"자동, DHCP 전용" this means Auto, for DHCP. (I guess "Auto" has same meaning with "Dynamic".

[image]
[image]

[Fifth Picture : when we operate the wireshark to check the connection.]
[image]

ChicagoPARK, 2021-05-07:
I also have different question.

I wanna ask about main purposes of Hub.
As I understood about Hub, the purpose of it is just connect multiple LiDARs at once.

However I think hub is quite pricey to have just that purpose. So I guess it has critical function. (Such as calibrate all the pointcloud data from multiple lidars automatically)

I also heard that it is possible to control by using Livox viewer and control it. But if I use Ubuntu 18.04 which is not supported by Livox viewer, Do you think there is no benefit to use Hub?

Thank you for reading my question.
Looking for your feedback.

Livox-SDK, 2021-05-13:
Hi, ChicagoPARK
     The Livox viewer support Ubuntu 18.04!

ChicagoPARK, 2021-05-13:
Appreciate for your response!
Can I get the web path??
Cause from Livox page there are just window and 16.04 version..

ChicagoPARK, 2021-05-13:
a.... You mean it is possible to install 16.04 on 18.04 ...

Got it

Livox-SDK, 2021-05-13:
Yes, you can install Livox viewer on ubuntu 18.04.

ChicagoPARK, 2021-05-13:
Thanks Thanks.
Your helps make us can proceed to the next progress.

It gonna be the last question..
After I use the Livox hub, I think the main function of hub is that having a good match with Livox viewer.
So it was not necessary to code additionally for merge all the each Livox Lidars.

I wanna ask is there any other good functions of Hub which we did not recognize?

Livox-SDK, 2021-05-25:
Another good function of Hub is timestamp synchronization. Hub can easily synchronizing the timestamp when multiple LiDAR connected.

ChicagoPARK, 2021-05-25:
Thank you for ur response!
We've used hub till now. And I felt it is quite amazing!
Your answer works lot hoho

Have a nice day there

shettygr, 2023-12-11:
hi guys, iam using the livox mid-360 and unable to detect in the livox viewer is there any help from your side