Livox-SDK/livox_ros_driver #82: How to get the point rate with livox_lidar_rviz.launch

UchiyKei1320, 2021-04-05:
Thank you for your help.

I try getting the point rate with this command "roslaucn livox_ros_driver livox_lidar_rviz.launch" on Livox Mid 70.

I am very afraid of asking this kind of basic question but, what should I get the the point rate on display_lidar_points.rviz view.

On spec, Mid 70 can probably have 100,000 points per seconds.

Livox-SDK, 2021-04-21:
Hi UchiyKei1320,

I am sorry to say that, but for now, the exact point number is not able to be accessed directly. File size may give you a brief idea of how many points it may contains.

UchiyKei1320, 2021-04-30:
Thank you for your answer.

I will try to calculate the number of points from the file size.

joonr7, 2021-08-17:
Please do this:
```bash
$ rostopic hz /livox/lidar
```