Livox-SDK/livox_ros_driver #83: 使能multi_topic导致CPU负载300%

carlyangdong, 2021-04-12:
commit：1d6e869a3e7422669825da51fc788f0128f141a1
设备：两个mid70
平台：Nvidia Xavier AGX
使用launch：livox_lidar.launch
multi_topic: 0 时cpu使用正常，但我们想把两个mid70的topic分开，因此将multi_topic置1，驱动运行几秒钟之后，livox驱动进程的cpu使用达到300%。

Livox-SDK, 2021-04-12:
carlyangdong，
你好，发送频率设置的是多少？建议先在 X86 平台上验证下，正常情况下不会有这么差别

carlyangdong, 2021-04-15:
只是使能multi_topic，其他配置都是默认的，频率10Hz。
我在x86上也测试过了，也有概率出现CPU 300%的情况，x86和arm64都是概率出现，但概率挺大的。
这边无法贴图片，我只能文字描述下大致状态。运行一会突然就CPU使用飙升到300%稳定，top查看有3个子线程CPU 100%，这3个核每个cpu的sy占60-70%，us占30-40%，表明子线程一直在yield触发内核调度，问题可能处在sdk，我这边使用的是sdk v2.3.0。

Livox-SDK, 2021-04-19:
carlyangdong, 
       你好！
我们先看下是否能复现这个问题，然后再来分析原因。
你那边如果有什么进展或者进一步非发现，请更新到这里，谢谢!

carlyangdong, 2021-04-19:
[image]

russellyq, 2021-06-03:
您好，请问您是怎么launch多个lidar的呢？

我连接多个lidar之后始终只能launch其中一个。

谢谢。

carlyangdong, 2021-06-19:
> 您好，请问您是怎么launch多个lidar的呢？
> 
> 我连接多个lidar之后始终只能launch其中一个。
> 
> 谢谢。

ros驱动默认是可以自动连接多个livox的，自发现是sdk中实现的。2个livox的topic同一个，frame_id也一样，频率翻倍为20Hz。

chengnanx, 2025-02-25:
解决了吗