Livox-SDK/livox_ros_driver #84: 关于PTP时间同步的说明（explanation about PTP time synchronization）

borongyuan, 2021-04-30:
I've explaned this in https://github.com/Livox-SDK/livox_ros_driver/pull/2#issuecomment-796532241 and https://github.com/Livox-SDK/livox_ros_driver/pull/2#issuecomment-830006257. But lots of developers are facing this problem and even try to modify the code in a wrong way. So I add some explanation here in chinese.
根据官方文档，Livox支持3种同步方式。其中PTP同步并不由驱动来管理，而由主机的PTP服务来管理。ROS包中只有GPS同步并不意味着PTP同步不被支持。只要正确配置PTP服务，在使用ROS包时便能获得正确的时间戳。在Linux系统中，ptpd和linuxptp均可提供PTP服务，任选一个即可。都可以使用apt安装，无需编译。可以通过init.d或systemd在开机时自启。
若使用linuxptp，需同时开启ptp4l和phc2sys。ptp4l用于同步livox和硬件时钟，phc2sys用于同步硬件时钟和系统时钟。因此同时使用两者才能让Livox的时钟与系统同步。若使用ptpd，则会自动同步到系统时钟。
因为比较懒，就先不发PR改文档了。总之别乱改代码！！！

msadowski, 2021-04-30:
Many thanks for the insights on PTP (translated it with google translate, your description matches my expectations).

If someone wants some step-by-step instructions on ptp setup then [Ouster software integration manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf) describes it quite well in their software manual on page 85.

As for Livox integration @borongyuan I have one question. With my ptp setup I'm seeing an average delay of 0.370 seconds (tested with `rostopic delay /livox/lidar`) for my MID-40. Does this number match your experience with these units? I'm trying to build an open-source proof of concept precise mapping module so having precise information about the delay is very important for me.

borongyuan, 2021-05-07:
Have you compared the delay with NTP service ON/OFF?

gongyue666, 2021-12-07:
你好，我根据PTP教程设置了硬件同步的系统时间，利用livox_viewer软件查看也是1588，但/livox/lidar输出的消息时间戳和其他节点输出的始终不一样。具体描述可以看这里https://forum.livoxtech.com/thread-206-1-1.html