Livox-SDK/livox_ros_driver #87: How to launch two devices

russellyq, 2021-06-01:
Hi, Thanks for your great work

I am trying to launch two devices at same time.

I am using "roslaunch livox_ros_driver livox_lidar.launch multi_topic:="1" bd_list:="XX&XX""

However, it can only launch one of these two devices.

russellyq, 2021-06-02:
To be clear, instead of using livox hub, are there any ways to launch two devices which are directly connected to laptop ?

Livox-SDK, 2021-06-04:
"roslaunch livox_ros_driver livox_lidar.launch multi_topic:="1" bd_list:="XX&XX""

"multi_topic commandline parameter " is  depreated, 

add the broadcast code to livox_ros_driver/livox_ros_driver/config/livox_lidar_config.json file, for example:

{
"lidar_config": [
{
"broadcast_code": "1PQDH5B00100041",
"enable_connect": true,
"return_mode": 0,
"coordinate": 0,
"imu_rate": 0,
"extrinsic_parameter_source": 0,
"enable_high_sensitivity": false
},
{
"broadcast_code": "0TFDG3U99101431",
"enable_connect": true,
"return_mode": 0,
"coordinate": 0,
"imu_rate": 0,
"extrinsic_parameter_source": 0,
"enable_high_sensitivity": false
}
],

and then,use the following command :
roslaunch livox_ros_driver livox_lidar.launch

russellyq, 2021-06-04:
Thanks for your reply.

I changed  the config file as follows:

    "lidar_config": [
        {
            "broadcast_code": "3WEDH7600103381",
            "enable_connect": true,
            "enable_fan": true,
            "return_mode": 0,
            "coordinate": 0,
            "imu_rate": 0,
            "extrinsic_parameter_source": 0,
            "enable_high_sensitivity": false
        },
        {
            "broadcast_code": "1HDDH1200100801",
            "enable_connect": true,
            "enable_fan": true,
            "return_mode": 0,
            "coordinate": 0,
            "imu_rate": 0,
            "extrinsic_parameter_source": 0,
            "enable_high_sensitivity": false
        }
    ],

However, it still can only launch one of these two lidars.

@Livox-SDK

Livox-SDK, 2021-06-04:
Hi, russellyq
     Both lidars must be connected to your PC in the same loacal erea network！
     You can buy a network switch or network router.

russellyq, 2021-06-11:
Thank you ! That works !
@Livox-SDK

closed 2021-06-11 by russellyq