Livox-SDK/livox_ros_driver #88: Connection has losg

Zhigal, 2021-06-09:
Hello, we are running livox horizon and in a few hours connection gets lost and does not recover. We found only one way how to fix this, wa have to replug lidar ethernet cable from the router and then connection recovers immediatly.
Any ideas what is wrong and what we may check? 
Thanks in advance.

Log:

> May 07 14:33:05 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:05 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:05 ox01 bash[19771]: [2021-05-07 14:33:05.977] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:05 ox01 bash[19771]: [2021-05-07 14:33:05.977] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:05 ox01 bash[19771]: [2021-05-07 14:33:05.977] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:05 ox01 bash[19771]: [2021-05-07 14:33:05.977] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] New Device  [device_discovery.cpp] [OnData] [99]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] Handle: 0  [device_discovery.cpp] [OnData] [100]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] Broadcast Code: 1HDDH3200105041  [device_discovery.cpp] [OnData] [101]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] Type: 3  [device_discovery.cpp] [OnData] [102]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] IP: 192.168.1.102  [device_discovery.cpp] [OnData] [103]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] Command Port: 55501  [device_discovery.cpp] [OnData] [104]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info] Data Port: 56001  [device_discovery.cpp] [OnData] [105]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.001] [console] [info]  Send Command: Set 0 Id 2 Seq 3  [command_channel.cpp] [Send] [243]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.026] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 3  [command_handler.cpp] [OnCommand] [118]
May 07 14:33:06 ox01 bash[19771]: Lidar[1HDDH3200105041] status_code[0] working state[5] feature[0]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.026] [console] [info]  Send Command: Set 0 Id 2 Seq 4  [command_channel.cpp] [Send] [243]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.528] [console] [warning] Command Timeout: Set 0, Id 2, Seq 4  [command_channel.cpp] [OnTimer] [142]
May 07 14:33:06 ox01 bash[19771]: [2021-05-07 14:33:06.528] [console] [info]  Recieve Ack: Set 0 Id 2 Seq 4  [command_handler.cpp] [OnCommand] [118]
May 07 14:33:09 ox01 bash[19771]: Device Query Informations Failed : -4
May 07 14:33:09 ox01 bash[19771]: [2021-05-07 14:33:09.035] [console] [info]  Device 0 removed   [device_manager.cpp] [RemoveDevice] [97]
May 07 14:33:10 ox01 bash[19771]: Lidar[1HDDH3200105041] disconnect!
May 07 14:33:10 ox01 bash[19771]: [2021-05-07 14:33:10.081] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]
May 07 14:33:10 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:10 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:10 ox01 bash[19771]: [2021-05-07 14:33:10.081] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:10 ox01 bash[19771]: [2021-05-07 14:33:10.081] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:10 ox01 bash[19771]: [2021-05-07 14:33:10.081] [console] [info] Command Port: 55502  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:10 ox01 bash[19771]: [2021-05-07 14:33:10.081] [console] [info] Data Port: 56002  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:11 ox01 bash[19771]: [2021-05-07 14:33:11.089] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]
May 07 14:33:11 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:11 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:11 ox01 bash[19771]: [2021-05-07 14:33:11.090] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:11 ox01 bash[19771]: [2021-05-07 14:33:11.090] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:11 ox01 bash[19771]: [2021-05-07 14:33:11.090] [console] [info] Command Port: 55503  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:11 ox01 bash[19771]: [2021-05-07 14:33:11.090] [console] [info] Data Port: 56003  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:12 ox01 bash[19771]: [2021-05-07 14:33:12.097] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]
May 07 14:33:12 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:12 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:12 ox01 bash[19771]: [2021-05-07 14:33:12.097] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:12 ox01 bash[19771]: [2021-05-07 14:33:12.097] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:12 ox01 bash[19771]: [2021-05-07 14:33:12.097] [console] [info] Command Port: 55504  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:12 ox01 bash[19771]: [2021-05-07 14:33:12.097] [console] [info] Data Port: 56004  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:13 ox01 bash[19771]: [2021-05-07 14:33:13.105] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]
May 07 14:33:13 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:13 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:13 ox01 bash[19771]: [2021-05-07 14:33:13.105] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:13 ox01 bash[19771]: [2021-05-07 14:33:13.105] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:13 ox01 bash[19771]: [2021-05-07 14:33:13.105] [console] [info] Command Port: 55505  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:13 ox01 bash[19771]: [2021-05-07 14:33:13.105] [console] [info] Data Port: 56005  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:14 ox01 bash[19771]: [2021-05-07 14:33:14.113] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]
May 07 14:33:14 ox01 bash[19771]: In automatic connection mode, will connect 1HDDH3200105041
May 07 14:33:14 ox01 bash[19771]: Could not find raw config, set config to default!
May 07 14:33:14 ox01 bash[19771]: [2021-05-07 14:33:14.113] [console] [info] LocalIP: 192.168.1.2  [device_discovery.cpp] [OnBroadcast] [201]
May 07 14:33:14 ox01 bash[19771]: [2021-05-07 14:33:14.113] [console] [info] DeviceIP: 192.168.1.102  [device_discovery.cpp] [OnBroadcast] [202]
May 07 14:33:14 ox01 bash[19771]: [2021-05-07 14:33:14.113] [console] [info] Command Port: 55506  [device_discovery.cpp] [OnBroadcast] [203]
May 07 14:33:14 ox01 bash[19771]: [2021-05-07 14:33:14.113] [console] [info] Data Port: 56006  [device_discovery.cpp] [OnBroadcast] [204]
May 07 14:33:15 ox01 bash[19771]: [2021-05-07 14:33:15.121] [console] [info]  Broadcast broadcast code: 1HDDH3200105041  [device_discovery.cpp] [OnBroadcast] [150]

Rrycbarm, 2021-12-02:
Did you managed to solve?

Zhigal, 2021-12-02:
@Rrycbarm some kind of...
We added a switch to disable a power for a few seconds and reenable it back again.

closed 2022-07-12 by Zhigal

85256638, 2025-06-21:
> [@Rrycbarm](https://github.com/Rrycbarm) some kind of... We added a switch to disable a power for a few seconds and reenable it back again.

@Zhigal , Hi can you let me know what kind of switch do you use to turn off the power?