Livox-SDK/livox_ros_driver #91: Set GPRMC synchronization time error code: -1.

meng-zha, 2021-07-01:
- 平台：Nvidia Jetson Xavier

- 激光雷达：Livox Mid100
- GPS模块：U-blox 6

- 问题：使用GPS时间同步失败，debug发现RMC的packet读取有问题，不能准确读取到GPRMC的信息。举例如下：
`rmc:$GPGGA,085630.00,4000.00291,N,11619.62728,E,2,07,1.92,93.4,M,-8.8,M,,0000*70
$GPGSA,A,3,06,07,02,20,50,15,13,,,,,,3.32,1.92,2.70*0D
Set GPRMC synchronization time error code: -1.
Set GPRMC synchronization time error code: -1.
Set GPRMC synchronization time error code: -1.`
其中rmc输出为[ReceiveSyncTimeCallback函数的参数](https://github.com/Livox-SDK/livox_ros_driver/blob/master/livox_ros_driver/livox_ros_driver/lds_lidar.cpp#L538)，理论上应该为每次pps的GPRMC值。怀疑common代码下的串口解析有问题。

- 其他信息：利用串口工具读取的GPS信息正常。
在Windows平台使用Livox SDK同步采集正常。

- 详细log: [GPS输出信息](https://cloud.tsinghua.edu.cn/f/efeea9e92130400fa065/)、
[roslaunch livox_lidar.launch 输出信息](https://cloud.tsinghua.edu.cn/f/50e1ca3a255b4ddcbaac/)

Livox-SDK, 2021-07-06:
抱歉，Livox Mid100 不支持该功能

meng-zha, 2021-07-06:
> 抱歉，Livox Mid100 不支持该功能

 @Livox-SDK 请您说明白“不支持”的是什么功能。正如之前说的，在windows平台使用gps同步mid100是成功的，这个明显只是livox_ros_driver中代码的bug或者使用的时候配置的问题，而不是一句简单的“不支持”。mid100其实就是三个同步的mid40，要是mid100不支持gps同步，那意思是mid40也不支持了？

希望回复issue的时候能说明清楚。

meng-zha, 2021-07-10:
通过把这个repo的TimeSync模块替换成Livox-SDK中的synchro模块，解决的GPS的bug问题。如果后续有遇到GPS解码失败的问题，可以参考https://github.com/meng-zha/livox_ros_driver

Livox-SDK, 2021-07-10:
meng-zha,
      不好意思，给您造成了困惑！
      livox ros driver 中 GPS 协议解包部分，是一个 GPS 协议的解包框架，只要满足 GPS 协议格式（即‘$’开头，‘*’结尾，同时满足校验），都可以正确解析处理，例如：$GPGGA,$GPRMC,$GPGSV等等；
      您的报错，应该是非 $GPRMC 格式的数据导致的；
      如果是这样，即使出现报错，也不影响实际的时间同步；
[image]