Livox-SDK/livox_ros_driver #93: 点云格式

huduo0812, 2021-07-09:
我现在同时需要发布pointcloud2和Livox customized pointcloud format这两种格式的点云，应该怎么修改代码呢

Livox-SDK, 2021-07-09:
Hi huduo0812,

我建议您按照两种格式的差异写一段格式转换的代码；

huduo0812, 2021-07-09:
> Hi huduo0812,
> 
> 我建议您按照两种格式的差异写一段格式转换的代码；

我们也试过，以我们现在的技术可能写不出来，但是我试过更改你们的代码同时发布两种格式的点云，但是自己改的代码发布的pointcloud2类型的点云显示出来貌似只有一线，特别稀疏，可能是什么原因呢？
[image]
[image]
[image]
[image]
这是我们自己修改的地方，你们可以出一个Livox customized pointcloud format转成pointcloud2格式的功能包吗，非常感谢

Livox-SDK, 2021-07-10:
建议不要同时发送两种点云格式的数据包，能说一下你的原始需求吗？

huduo0812, 2021-07-11:
> 建议不要同时发送两种点云格式的数据包，能说一下你的原始需求吗？

我想用avia做导航和避障，但是我没有里程计信息，需要用avia和imu获取，fast_lio可以输出里程计信息，但是fast_lio需要的点云格式是Livox customized pointcloud，同时还需要point cloud2格式的点云转换成scan类型的激光数据避障。所以需要两种点云格式的数据包

Livox-SDK, 2021-07-12:
可以使用point cloud2格式，然后将 fast_lio 的输入点云格式适配成 pointcloud2？

Livox-SDK, 2021-07-12:
[image]

huduo0812, 2021-10-20:
> @huduo0812 您好，请问您解决这个问题了吗？是通过将CustomMsg转为PointCloud2，还是改了fast_lio的代码呢？我也在做类似的任务，期待解答~

没有解决，自己试过同时发两种格式的，但是效果不理想。可以加个联系方式交流嘛？

zhaomeihu, 2021-10-21:
> 没有解决，自己试过同时发两种格式的，但是效果不理想。可以加个联系方式交流嘛？

我的邮箱1442703432@qq.com

maxpolzin, 2022-12-13:
I have a ROS bag with Livox Lidar data recorded in the PointCloud2 format. 
However, I need my data in Livox's CustomMsg format. 
Did you figure out how to convert between the two formats?

I appreciate your inputs.

1010186, 2022-12-15:
大佬知道这个怎么解决吗，运行roslaunch livox_ros_driver livox_lidar_msg.launch，想获得一个customMsg格式的bag包，但是出现
[ WARN] [1671087467.186709118]: Shutdown request received.
[ WARN] [1671087467.186750218]: Reason given for shutdown: [[/livox_lidar_publisher] Reason: new node registered with same name]
.
.
.
[ERROR] [1671087471.099274355]: Client [/rviz] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.

panchenwei, 2024-05-13:
我也是，请问怎么在humble中录制customMsg的bag呢