Livox-SDK/livox_ros_driver #94: Where is the bag path?

ManChrys, 2021-07-17:
Lidar : Livox mid 40
OS : Ubuntu 18.04

Hello guys,

i have a question. When i use livox_lidar.launch file i enable the "rosbag_enable" option. 

~~~
<node pkg="rosbag" type="record" name="record" output="screen"
          		args="-a"/>
~~~

In the end of the file i saw the line above, so if i recard all topics, because we use -a option, where is the bag file saved? I cant find the path. Should i set the path inside the launch file? and if yes how ?

Thanks in advance.

Livox-SDK, 2021-07-22:
Hi,ManChrys
    you can append the path to "-a", this parameter item is not ofen used.

Livox-SDK, 2021-07-22:
you can use the "rosbag record" command in separate terminal

ManChrys, 2021-07-23:
yy that's actually what I did!!

Thank you guys.

closed 2021-07-23 by ManChrys