Livox-SDK/livox_ros_driver #95: Rosbag to PointCLoud

ManChrys, 2021-07-21:
hello,

i would like to ask you how can i convert rosbag to a ply format for example las,laz,ply

i saw that page http://wiki.ros.org/pcl_ros#pointcloud_to_pcd but when im converting bag to pcd it creates a lot of pcds and not one pcd. It is one pcd for every frame. 

Any suggestions ? Thanks in advance.

Livox-SDK, 2021-07-23:
Hi ManChrys,

There is a part of code for saving the pointcloud as pcd in the code livox_mapping which can be refered:
[image]

ManChrys, 2021-07-23:
Ok I see.

Thank you a lot.

ManChrys, 2021-07-24:
hello,

i tried livox mapping and the result of pcd has big difference compared to bag file. Is there something else to try ? I have to convert it to a common pointcloud format like ply, las, laz so i can import them to point cloud softwares like cloudcompare. 

livox_mapping pcd:

[image]

rosbag (converted from .lvx file)

[image]

closed 2023-08-31 by ManChrys

Martinyllim, 2026-09-07:
bag_to_pcd from pcl_ros is built to dump one PCD per frame since it's just writing out each PointCloud2 message as it comes, it's not doing any registration or accumulation between frames, so that behavior is expected rather than a bug. To get one unified cloud (and LAS/LAZ/PLY from there) you need something doing scan matching or SLAM to merge the frames into a common reference frame first, then you can export the merged result in whatever format you want. pcl_ros's concatenate_and_downsample node can crudely stack the individual PCDs into one file if the platform barely moved, but if there's any real motion in your recording you'll get ghosting and doubled geometry without actual registration between scans.

If you're looking for something that handles the registration step, GLIM or FAST-LIO2 are common picks for Livox specifically. Full disclosure, I built a browser tool called LidarFlow that runs GLIM on a rosbag or MCAP and hands back PCD, PLY, and LAZ from the merged result, might save you standing up the SLAM pipeline yourself if you just want the output.