Livox-SDK/livox_ros_driver #97: How to merge multi Lidars PointCloud connected with hub?

h-wata, 2021-08-05:
Hi,
I try to connect 5 Horizon to Livox Hub and saved .lvx file.
After that lvx convert to bag file by lvx_to_rosbag.launch.

**Configuration**
- timesync: ptp
- lvx_to_rosbag.launch default configuration
    -  publish_freq = 10

**livox viewer(lvx)**
[image]
 
I confused that `/livox/lidar` topicc seems to each lidar pointcloud as follow,
**RVIZ**
[image]

```
rostopic hz /livox/lidar
subscribed to [/livox/lidar]
average rate: 54.471
        min: 0.000s max: 0.100s std dev: 0.03807s window: 50
average rate: 52.139
        min: 0.000s max: 0.100s std dev: 0.03877s window: 100
average rate: 51.392
        min: 0.000s max: 0.100s std dev: 0.03871s window: 150
```

Please tell me how to publish merged 5 Lidars pointcloud like .lvx file.

Livox-SDK, 2021-09-02:
Set Decay Time to 0.1, 0.2, 0.5, or 1, etc. 
The "Decay Time" must be  greater than 0.

h-wata, 2021-09-03:
> Set Decay Time to 0.1, 0.2, 0.5, or 1, etc.
> The "Decay Time" must be greater than 0.

What you mean is that one topic is published from only one lidar, right?
If so, I use the node to merge PointCloud2 topic like [points_concat_filter](https://github.com/Autoware-AI/core_perception/blob/master/points_preprocessor/launch/points_concat_filter.launch).

Thank you for reply.

Livox-SDK, 2021-09-03:
"What you mean is that one topic is published from only one lidar, right?"
yes，you can sync and merge all lidars when you receive enough topics!

h-wata, 2021-09-03:
Thanks for your support!

closed 2021-09-03 by h-wata