Livox-SDK/livox_ros_driver #105: timebase format

mohd109, 2021-09-27:
Hello everyone, 
First of all, I should thank the Livox team for all of their valuable efforts.
I want to connect LOAM and color assignment codes, so I have to synchronize different color frames and point cloud messages because the scanner is moving. In order to do so, I just need the correct timestamp based on the computer clock to match with externally captured images. I have a problem with the format of timebase. Can you describe how the timebase variable represents the time based on seconds? "livoxCloud.header.stamp.toSec()" will give me the relative time based on seconds, but I have to convert it to the system clock.

Thanks a lot.

closed 2021-09-27 by mohd109

kunnalparihar, 2021-10-19:
@mohd109 You got any solution to the above issue

mohd109, 2021-10-19:
Both timestamp and timebase are in nanoseconds. But I didn't find any absolute time value in the file, so you have to get the accurate clock at the beginning of the recording manually.