Livox-SDK/livox_ros_driver #107: livox_ros_driver/CustomMsg.h

yuchuxiang, 2021-10-15:
fatal error: livox_ros_driver/CustomMsg.h: No such file or directory
 #include <livox_ros_driver/CustomMsg.h>

缺少这个头文件是什么情况，在原程序包中也没找到这个livox_ros_driver/CustomMsg.h头文件，这个怎么解决？

flashlau, 2022-10-27:
在devel/include里面，不是源程序包的而是compile产生的

peibiao-zhang, 2022-11-08:
我也遇到了相同的问题，解决办法是删了build和devel文件夹，多编译几次

maxpolzin, 2022-12-13:
There is a dependency in the CMakeFile missing. 
Add this line to ensure the livox_ros_driver messages are compiled first. 
add_dependencies(r3live_LiDAR_front_end ${catkin_EXPORTED_TARGETS})

maxpolzin, 2023-01-20:
Hi @farhad-dalirani,
This means that the livox_ros_driver/CustomMsg is not compiled. 
Have you compiled the livox_ros_driver? 
Have you sourced your ros_ws afterwards, e.g. source ~/catkin_ws/devel/setup.bash?

farhad-dalirani, 2023-01-20:
Thank you, source ~/catkin_ws/devel/setup.bash did the job.

hipforth, 2023-11-02:
> There is a dependency in the CMakeFile missing. Add this line to ensure the livox_ros_driver messages are compiled first. add_dependencies(r3live_LiDAR_front_end ${catkin_EXPORTED_TARGETS})

this is useful

iason-r, 2024-02-01:
> There is a dependency in the CMakeFile missing. Add this line to ensure the livox_ros_driver messages are compiled first. add_dependencies(r3live_LiDAR_front_end ${catkin_EXPORTED_TARGETS})

I used your method but encountered a new problem

CMake Error at CTLO/CMakeLists.txt:22 (add_dependencies):
  add_dependencies called with incorrect number of arguments

It seems like there are some issues with "add_dependencies"
can you tell me the answer