Livox-SDK/livox_ros_driver #129: 运行livox_ros_driver 提示无法找到 [livox_ros_driver_node] ·「系统在安装ros后，又装了anaconda...」

Xu-Huitong, 2022-07-27:
运行：roslaunch livox_ros_driver livox_lidar_msg.launch 
... logging to /home/xuhuitong/.ros/log/ef12ded4-0d85-11ed-97fa-df185c5966bd/roslaunch-bjtu-13259.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bjtu:35485/

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 100000000000000
 * /data_src: 0
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /multi_topic: 0
 * /output_data_type: 0
 * /publish_freq: 10.0
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /user_config_path: /home/xuhuitong/w...
 * /xfer_format: 1

NODES
  /
    livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)

auto-starting new master
process[master]: started with pid [13282]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ef12ded4-0d85-11ed-97fa-df185c5966bd
process[rosout-1]: started with pid [13316]
started core service [/rosout]
ERROR: cannot launch node of type [livox_ros_driver/livox_ros_driver_node]: Cannot locate node of type [livox_ros_driver_node] in package [livox_ros_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)

Xu-Huitong, 2022-07-30:
已解决

kausalkyle, 2023-07-21:
How did you solve this @Xu-Huitong ?

85256638, 2024-11-15:
> 已解决

已解决，怎么解决的。。。。 @Xu-Huitong