Livox-SDK/livox_ros_driver #130: Deviced removed and reconnect

dariojavo, 2022-08-13:
Hi,

Hope you are doing well.

I am currently using the Livox MID-70 with a dhcp router configuration. While everything seems to be working, at some points while running the livox_lidar.launch file, I received the following message:

`[console] [info] Device 0 removed [device_manager.cpp][RemoveDevice] [97]`

and just after that 

`Lidar[#] disconnect!`

Which make me lose connection with the Lidar, and automatically after that it tries to connect again and keep going with the same issue forever. Is something that I am missing?

I also notice that the Command Port and Data Port are continuously changing, is that normal?

Hope you can give some guidelines of what could be the reason.

Best,
Dario

dariojavo, 2022-08-22:
Solved: Verify router manage dhcp.

closed 2022-08-22 by dariojavo

FengYuQ, 2023-10-22:
I still no idea how did you sloved it？Could you give more detail？