Livox-SDK/livox_ros_driver #134: 运行lvx_to_rosbag.launch出现REQUIRED process [livox_lidar_publisher-1] has died!

7216Inori, 2022-09-16:
在相似的issue（#77）中官方建议将imu_bag设为False，但仍然无法正常运行。该issue中的后续追问未能得到回答。

terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: ~/Desktop/***/lidardata/***.bag
================================================================================
REQUIRED process [livox_lidar_publisher-1] has died!
process has died [pid 15806, exit code -6, cmd /home/nvidia/livox/dirver/ws_livox/devel/lib/livox_ros_driver/livox_ros_driver_node 100000000000000 __name:=livox_lidar_publisher __log:=/home/nvidia/.ros/log/ba94ce66-355c-11ed-88d9-d8bbc1342065/livox_lidar_publisher-1.log].
log file: /home/nvidia/.ros/log/ba94ce66-355c-11ed-88d9-d8bbc1342065/livox_lidar_publisher-1*.log
Initiating shutdown!
================================================================================
[livox_lidar_publisher-1] killing on exit

PARAMETERS
 * /cmdline_str: 100000000000000
 * /data_src: 2
 * /enable_imu_bag: False
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /multi_topic: 0
 * /output_data_type: 1
 * /publish_freq: 10.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /xfer_format: 0

OS：Ubuntu 18.04
livox_ros_driver：2.6/2.5均无法成功

在有相似issue的情况下，提交了新的issue，十分抱歉。

7216Inori, 2022-12-04:
已解决

1010186, 2022-12-14:
> 已解决

你好，请问一下怎么解决的