Livox-SDK/livox_ros_driver #141: compile issue

66695, 2022-11-01:
i compiled it in ubuntu20.04 with ros noetic, and with problem:
/home/dfs/ws_livox/src/livox_ros_driver/livox_ros_driver/lddc.cpp:343:31:   required from here
/opt/ros/noetic/include/ros/message_traits.h:125:14: error: ‘const class std::shared_ptr<pcl::PointCloud<pcl::PointXYZI> >’ has no member named ‘__getMD5Sum’
  125 |     return m.__getMD5Sum().c_str();
      |            ~~^~~~~~~~~~~
......
/opt/ros/noetic/include/ros/message_traits.h:142:14: error: ‘const class std::shared_ptr<pcl::PointCloud<pcl::PointXYZI> >’ has no member named ‘__getDataType’
  142 |     return m.__getDataType().c_str();

66695, 2022-11-01:
https://github.com/Livox-SDK/livox_ros_driver/pull/110 has mentioned.

closed 2022-11-01 by 66695

HernandoR, 2024-12-23:
any fix concluded?